jdibenes/hl2ss #2: Port 3816 timeout error

GLARKI, 2022-11-17:
Dear authors and fellow GitHub colleagues,

I keep getting a time out error 10060 whenever I use the TCP Port 3816 that allows the client to send messages to a Unity Application.

The repo part regarding the viewer works as a charm so far so I figure it's me, but I'd still like to know if other people encounter this and have insights for me?

Trying to use the unity plugin as we speak. Would love to use the sensor data in my own projects as well.

If it's me close this issue and sorry for bothering.

jdibenes, 2022-11-19:
Hello @GLARKI 
Are you using the plugin DLL in a Unity App?
The application installed with the appxbundle does not support port 3816.

GLARKI, 2022-11-20:
Thanks for replying! Your repo is great for capturing and showing sensor data!

1.I'm trying to figure out how I can use this (raw) sensor data within my own HL2 application. As I'm new to research mode, I thought your plugin is a good start.

I do not see a .dll file mentioned in the readme or explorer. I can find it in the Reference you mention (HL2forCV). Although I'm not sure how to correctly implement it?

2. In your section "remote unity scene" you refer to the viewer and the TCP connection. In the "python client" section you specify which examples can be used. AKA it was my mistake, but thanks for clarifying.  
The port 3816 should thus be used with the unity plugin and not with the appxbundle from the repo, my bad.