jdibenes/hl2ss #3: hl2ss.winmd does not exist

srafaelm, 2022-11-21:
I followed the steps mentioned in the readme file however when trying to build the release ARM64 I keep getting this error.  I have installed the appxbundle on hololens, enable Research mode as well as Developer Mode. I have given permission for the app to access the camera, microphone, etc. My hololens were already paired due to previous apps.

 Has this ever happened to you? 

- I am using Visual Studio 2022 17.4.1

[image]

srafaelm, 2022-11-26:
Thank you, it solved the problem!! Great Work! I have been trying to stream all the Hololens cameras for some time. Is there a way to change the frame ratio of the cameras to have a more stable stream if the internet connection is weak?

jdibenes, 2022-11-26:
No, only the front camera is configurable. You can try reducing the bitrate of the streams.

closed 2023-01-19 by srafaelm