jdibenes/hl2ss #4: PV extrinsic calibration "Connection reset by peer" Error

yzhao520, 2022-11-22:
Hi,

I'm trying to get the pv extrinsic matrix using [pv_extrinsic_calibration.py](https://github.com/jdibenes/hl2ss/blob/main/tools/pv_extrinsic_calibration.py). But I run into this error named ConnectionResetError: [Errno 104] Connection reset by peer in [L42](https://github.com/jdibenes/hl2ss/blob/main/tools/pv_extrinsic_calibration.py#L42). Could you help me fix this issue? 

Btw, I'm able to run client_pv.py and some other streaming scripts in the viewer folder without any problem. So I'm not quite sure what's wrong.

[image]

Thanks!

jdibenes, 2022-11-23:
Hello @yzhao520 
Some of the PV tools were broken.
The latest commit should fix this issue.

yzhao520, 2022-11-23:
Thanks for the reply! Now it's working.

closed 2022-11-23 by yzhao520