jdibenes/hl2ss #5: Streaming Multiple Sensors Example Request

yzhao520, 2022-11-23:
Hi,

I'm trying to stream data from multiple sensors simultaneously. I have followed the examples shown in the viewer folder to stream data from HoloLens2. However, I found most of them only able to retrieve 1 or at most 2 sensors. In my case, I want to extract frames from pv camera, and also the corresponding frames from 4 grayscale cameras and depth. Could you maybe provide an example script?

I have wrote a script based the hl2ss_mp and hl2ss_utility before update. Here is an updated version following the most recent commits. Could you help me check whether this is best practice since I feel like I shouldn't start a separate pv client? In my case, if I combine everything in the producer and consumer, the program will just freeze for some reason. 

`

    hl2ss.start_subsystem_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO)

    enable = True

    def on_press(key):
        global enable
        enable = key != keyboard.Key.esc
        return enable

    listener = keyboard.Listener(on_press=on_press)
    listener.start()

    producer = hl2ss_mp.producer()
    producer.configure_rm_vlc(True, host, port1, hl2ss.ChunkSize.RM_VLC, hl2ss.StreamMode.MODE_1, profile, vlc_bitrate)
    producer.configure_rm_vlc(True, host, port2, hl2ss.ChunkSize.RM_VLC, hl2ss.StreamMode.MODE_1, profile, vlc_bitrate)
    producer.configure_rm_vlc(True, host, port3, hl2ss.ChunkSize.RM_VLC, hl2ss.StreamMode.MODE_1, profile, vlc_bitrate)
    producer.configure_rm_vlc(True, host, port4, hl2ss.ChunkSize.RM_VLC, hl2ss.StreamMode.MODE_1, profile, vlc_bitrate)
    producer.configure_rm_depth_longthrow(True, host, hl2ss.StreamPort.RM_DEPTH_LONGTHROW, hl2ss.ChunkSize.RM_DEPTH_LONGTHROW, hl2ss.StreamMode.MODE_1, hl2ss.PngFilterMode.Paeth)    
    
    producer.initialize(port1, hl2ss.Parameters_RM_VLC.FPS * buffer_length)
    producer.initialize(port2, hl2ss.Parameters_RM_VLC.FPS * buffer_length)
    producer.initialize(port3, hl2ss.Parameters_RM_VLC.FPS * buffer_length)
    producer.initialize(port4, hl2ss.Parameters_RM_VLC.FPS * buffer_length)
    producer.initialize(hl2ss.StreamPort.RM_DEPTH_LONGTHROW, hl2ss.Parameters_RM_DEPTH_LONGTHROW.FPS * buffer_length)
    
    producer.start(port1)
    producer.start(port2)
    producer.start(port3)
    producer.start(port4)
    producer.start(hl2ss.StreamPort.RM_DEPTH_LONGTHROW)

    manager = mp.Manager()
    # consumer = hl2ss_utilities.consumer()
    consumer = hl2ss_mp.consumer()
    sink_vlc1 = consumer.create_sink(producer, port1, manager, None)
    sink_vlc2 = consumer.create_sink(producer, port2, manager, None)
    sink_vlc3 = consumer.create_sink(producer, port3, manager, None)
    sink_vlc4 = consumer.create_sink(producer, port4, manager, None)
    sink_depth = consumer.create_sink(producer, hl2ss.StreamPort.RM_DEPTH_LONGTHROW, manager, ...)

    sinks = [sink_vlc1, sink_vlc2, sink_vlc3, sink_vlc4, sink_depth]
    [sink.get_attach_response() for sink in sinks]

    # pv camera
    client_pv = hl2ss.rx_decoded_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO, hl2ss.ChunkSize.PERSONAL_VIDEO, hl2ss.StreamMode.MODE_1, width, height, framerate, profile, pv_bitrate, 'bgr24')
    client_pv.open()

    while (enable):
        data_pv = client_pv.get_next_packet()

        _, data_vlc1 = sink_vlc1.get_nearest(data_pv.timestamp)
        _, data_vlc2 = sink_vlc2.get_nearest(data_pv.timestamp)
        _, data_vlc3 = sink_vlc3.get_nearest(data_pv.timestamp)
        _, data_vlc4 = sink_vlc4.get_nearest(data_pv.timestamp)

        _, data_depth = sink_depth.get_nearest(data_pv.timestamp) # payload.depth --> ndarray, uint16
`

jdibenes, 2022-11-24:
Hi,
We have added the script we used for the multi-camera demo: https://github.com/jdibenes/hl2ss/blob/main/viewer/demo_video.py.

yzhao520, 2022-11-26:
Thanks for the example! This is really helpful. Just curious, in this [demo_video](https://github.com/jdibenes/hl2ss/blob/main/viewer/demo_video.py) example, looks like there is no guarantee all frames are synchronized, right? If so, what's the best practice to synchronize all sensors? Should we have a main stream and call other streams with get_nearest function? Alternatively, I guess we could also check the timestamp and postprocess frames with closest timestamp?

yzhao520, 2022-11-28:
Another quick question: could you provide another example of generating rgbd pairs with pv camera and ahat depth sensor (similar to [open3d_rgbd_pv.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/open3d_rgbd_pv.py))? The problem I'm having now is that I found the ahat depth from data streams are in **uint8** format (instead of uint16 in long-throw mode) while the scale of the depth is 1000. So the depth maps seem not being captured in the correct range (0 - 0.255m?). Please let me know if I did anything wrong, thanks!

jdibenes, 2022-12-04:
> Thanks for the example! This is really helpful. Just curious, in this [demo_video](https://github.com/jdibenes/hl2ss/blob/main/viewer/demo_video.py) example, looks like there is no guarantee all [...]

That's right. Currently, the best way we have to synchronize data from different streams is get_nearest, which returns the data frame closest in time to the input timestamp.

> Another quick question: could you provide another example of generating rgbd pairs with pv camera and ahat depth sensor (similar to [open3d_rgbd_pv.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/open3d_rgbd_pv.py))? The problem I'm having now is that I found the ahat depth from data streams are in **uint8** format (instead of uint16 in long-throw mode) while the scale of the depth is 1000. So the depth maps seem not being captured in the correct range (0 - 0.255m?). Please let me know if I did anything wrong, thanks!
> 
> Btw, I could generate reasonable pairs with [open3d_rgbd_pv.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/open3d_rgbd_pv.py) in long-throw mode.

To stream ahat data at full framerate, we divide the depth by 4 to fit in uint8 format to pack as NV12 and encode. So, to recover the depth in meters, the depth values must be divided by 250. We have included an example using the front left grayscale camera and ahat to generate rgbd pairs [open3d_rgbd_ahat_vlc.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/open3d_rgbd_ahat_vlc.py). Our HoloLens has the latest OS version, so we cannot test PV+AHAT atm.

yzhao520, 2022-12-20:
Thanks for the explanations!

jdibenes, 2023-05-12:
Hi @vstadnytskyi 
It was renamed
https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_video.py

vstadnytskyi, 2023-05-12:
> Hi @vstadnytskyi It was renamed https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_video.py

Thank you for the quick response. And huge thanks for great library!