jdibenes/hl2ss #10: Port 3809 Timeout

mazrk7, 2023-01-25:
Hi, great repository and thanks for your work!

I'm encountering timeout troubles connecting to port 3809 via the plugin. All the other client scripts have worked flawlessly, but any scripts relying on establishing a connection for IPCPort.REMOTE_CONFIGURATION face the same issue.

For example, the `client_rc.py` script errors on a timeout connecting at the first instance of attempting to open the port in [this line](https://github.com/jdibenes/hl2ss/blob/main/viewer/client_rc.py#L61).

Any ideas on how to fix this? I can provide more information on my setup if required.

Thanks!

jdibenes, 2023-01-27:
Hi @mazrk7 
Are you initializing the plugin from the cpp code?
I noticed that in the readme, the `InitializeStreams(HL2SS_ENABLE_RM | HL2SS_ENABLE_MC | HL2SS_ENABLE_PV | HL2SS_ENABLE_SI)` does not include the `HL2SS_ENABLE_RC` flag, which is required to enable port 3809.

mazrk7, 2023-01-30:
Yes, perfect that fixed things! Thank you :)

closed 2023-01-30 by mazrk7