jdibenes/hl2ss #11: Externally Synchronized Timestamps

henrynoyes, 2023-02-02:
Hello, thanks for the great work you have done so far on this repository!

I was wondering if it is possible to generate data frame timestamps that are synchronized with an external reference such as UTC. From what I have read, the QPC timestamps given in each data frame do not have this property. If this is not possible with the current application, any advice on how to implement UTC-synchronized timestamps would be greatly appreciated.

Thank you!

jdibenes, 2023-02-03:
Hello @henrynoyes 
There is experimental support for converting QPC timestamps to [Windows FILETIME](https://learn.microsoft.com/en-us/windows/win32/api/minwinbase/ns-minwinbase-filetime).
Create a remote configuration client
`client = hl2ss.tx_rc(host, hl2ss.IPCPort.REMOTE_CONFIGURATION)`
then obtain the QPC to FILETIME offset as
`utc_offset = client.get_utc_offset(32)`
finally, add `utc_offset` to all QPC timestamps.

henrynoyes, 2023-02-03:
Ah, I see. This is helpful. Thank you!

closed 2023-02-03 by henrynoyes