jdibenes/hl2ss #12: Get distance from a point

AlexKanakis5, 2023-02-07:
Hello and thank you for your excellent work.

I have developed an application that uses the ZED camera's point cloud to get the distance from a specific point in the image. E.g., I get the pixel coordinates of a specific object and then I pass these coordinates to zed's point cloud. Can I achieve something similar using what you have developed here? Is there an example which you can provide?

jdibenes, 2023-02-09:
Hello @AlexKanakis5 
I'm not familiar with the ZED camera so I'm not sure I understand your question correctly.
Are you trying to find the intersection of the 3D ray (corresponding to the pixel coordinates of the object) with the point cloud to determine the distance between the camera and the object?
If so, this is possible using Open3D to do raycasting: [open3d_viewer_si.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/open3d_viewer_si.py) (lines 156-157 and lines 191-192, d is the distance).

RishaRanjan, 2023-03-08:
@jdibenes If I understand correctly, @AlexKanakis5 was asking to be able to, given a 2d point in the RGB image from the hololens, be able to find the corresponding depth information at that point. Is there something that exists in this package that allows us to do that?

AlexP-Coding, 2023-07-18:
I am also interested in the answer.
Also: did [open3d_viewer_si.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/open3d_viewer_si.py) change directories, or has it been merged with another file? I'm asking because the page currently shows a 404 error.