jdibenes/hl2ss #16: Stream to remote server

Birkenpapier, 2023-02-16:
Hello jdibenes,

thank you very much for your excellent work. The more I dig deeper into your project the more I am amazed by your skills.
I've understood that the current architecture is designed that the socket server is running on the HL and waiting for a connection to establish the stream. 
I want to change that behavior because I want to stream the data to a remote server and process it there. 

Mainly I am using the demo_2d_to_3d_segmentation.py which is unfortunately only working until the commit 9214fe3b2a. There were some changes inside the other util. files which are not compatible anymore. Maybe it is also worth to check it out and support it again with the newest version of your project? 

However, the HL is obviously not accesable from a different network so I want to know if there would be a suitable solution to stream the data to a remote server and then to a python client without making deep changes in your c++ project.

Is this possible, or should there be a WinSock client at the c++ application level to stream the data to the remote python socket server and then from there back to a python viewer of yours? 

BR,

Birkenpapier

jdibenes, 2023-02-18:
Hello @Birkenpapier 
The 2d-3d segmentation demo has been updated and should work now.
Some options that maybe could work without making changes to the c++ project:
- Port forwarding to expose the HL ports
- Have a computer on the same network as the HL forward the data to remote server
Otherwise, the code in the stream_x.cpp files would have to be adapted to stream the data to the remote sever, as you mentioned.

Birkenpapier, 2023-03-17:
Thank you very, very much for your answer.
I did resolve this issue by implementing a client creation function for the C++ application which starts to stream the long-throw and PV data immediately after the unity app starts.

closed 2023-03-17 by Birkenpapier