jdibenes/hl2ss #19: Does hl2ss support ARM

xw0810, 2023-02-21:
Hello jdibenes,

Thank you very much for your excellent work. Your work help us a lot as we are not familiar with hololens2. We recently want to integrate webrtc into the project, but it does not support ARM64. Does hl2ss support ARM? looking forward to your answer

jdibenes, 2023-02-27:
Hello,
Building hl2ss for ARM(32) is not supported. I'm not sure if the [Research Mode API](https://github.com/microsoft/HoloLens2ForCV) supports ARM yet. If it does, it might be possible to build hl2ss for ARM. If it does not, then it still might be possible to build a reduced version without the Research Mode sensors but this might not work for you. In either case, the functions in neon.cpp would have to be reimplemented for ARM.

Abdul-Mukit, 2023-03-13:
Hello, I think [microsoft psi](https://github.com/microsoft/psi) uses the research mode API to expose the sensors. We have used psi in ARM applications before so I think Research Mode API supports ARM.