jdibenes/hl2ss #24: Python interact with other Unity models

Enzo-Kerkhof, 2023-03-09:
Dear Juan Carlos Dibene,

First of all, thank you a lot for all the well written code and documentation. I've been able to run everything smoothly with your readme documents.
For my project, I would like to interact with preloaded objects/holograms already in the Unity app.
Would there be a method to find the key/id of the objects that are present in the Unity scene and be able to use all communication functions you have made as is?

Best,
Enzo

Enzo-Kerkhof, 2023-03-13:
For others trying something similar:

According to the Unity scripting API docs here: https://docs.unity3d.com/ScriptReference/Object.GetInstanceID.html
We can find all Instance IDs of available Game Objects using a script like:
`

    // Find all the game objects and display their Instance IDs
    void Start()
    {
        Object[] allObjects = Object.FindObjectsOfType<GameObject>();

        foreach (GameObject go in allObjects)
        {
            Debug.Log(go + " is an active object " + go.GetInstanceID());
        }
    }
`

In this repo inside the C# script in `/unity/hl2ss.cs` [lines 93-102](https://github.com/jdibenes/hl2ss/blob/775469f8f1618f24b20a253fcea40afee849f5a0/unity/hl2ss.cs#L93) a dictionary `m_remote_objects` is made where [all Instance IDs are added to when creating a new GameObject](https://github.com/jdibenes/hl2ss/blob/775469f8f1618f24b20a253fcea40afee849f5a0/unity/hl2ss.cs#L162).

Using the above lines from the Unity docs to find all Instance IDs and adding them to this dict will make it, probably, possible to interact with existing Unity objects. I'll update when this post in the case I get it to work.

Enzo-Kerkhof, 2023-03-16:
Managed to find a solution, will post the code later!

Enzo-Kerkhof, 2023-03-20:
To conclude:

If you make a public variable in `/unity/hlss.cs` you are able to drag a model to the C# script attached to the main camera in the Unity editor. If you then use this variable in the C# code to change the behaviour of the Create Primitive function to add this Unity InstanceID to the m_remote_objects dict, you can interact with this model from the python side. Please note to keep te code from adding the existing Game Object to the m_remote_objects dict over and over: to comment out the line to destroy the object from the remove function and use the remove function from the python side every time you interact with the model.

closed 2023-03-20 by Enzo-Kerkhof

kookoo9999, 2023-06-15:
hello, @Enzo-Kerkhof 
I saw your cooment in this page, thank you for sharing your experience.
I'm tryna interact with preloaded object, actually my goal is to control a transform of preloaded transform in python side.
But I coudn't understand above comment.
Can you tell me how I can do interact preloaded object in python side??
Thank you for reading.

kookoo9999, 2023-07-05:
@Enzo-Kerkhof 
Thank you for you answer but I'm sorry too late. 
First of all, I solve my problem. so I can interact with already exsits in the Unity deitor now.

I tryna calibrate with Unity world coordinate and Optical Tracking System(OTS) coordinate but it's difficult to calibrate .. 
If you had experience like my case, can you advice ?

environmet:
Unity - unit : M , left-handed.
OTS - unit : mm, right handed.
Thank you for reading.
I hope you have a nice day!

Civs17, 2023-09-06:
Sorry I have the same problem. Can you explain me how you did?

Enzo-Kerkhof, 2023-09-19:
@Civs17 Hi, which problem are you referring to?
The Unity to OTS coordinate system issue or the existing game object issue?

Civs17, 2023-09-19:
The Unity to OTS coordinate system issue