jdibenes/hl2ss #26: Pointcloud alignment inside unity application

Birkenpapier, 2023-03-17:
Hello,

thank you very much once again for creating and documenting such a nice project!
After tinkering with it around I want to visualize the results of the pointcloud directly inside the unity application.
Therefore I am streaming the pointcloud array over a custom made rest api back to the unity app.

Unfortunately I found out that the constructed pointcloud from open3d is not aligned with unitys coordinate system. 
It seems to be inversed because the points are placed in my observations mirrored. 

Do you have an idea or maybe any other developer how to align the reconstructed pointcloud from open3d inside unity to be placed at the same positions where they got recorded? 

[There is another developer which did exactly that](https://github.com/petergu684/HoloLens2-ResearchMode-Unity) but with unity version < 2021.2 because the methods for getting the unityWorldOrigin coordinatesystem got discontinued:
        IntPtr WorldOriginPtr = UnityEngine.XR.WSA.WorldManager.GetNativeISpatialCoordinateSystemPtr();
        unityWorldOrigin = Marshal.GetObjectForIUnknown(WorldOriginPtr) as Windows.Perception.Spatial.SpatialCoordinateSystem;

I would appreciate any help and thankt you very, very much.

BR,

Birkenpapier

Enzo-Kerkhof, 2023-03-20:
> It seems to be inversed because the points are placed in my observations mirrored.
Yes, Unity's coordinate system is left-handed and Open3D + OpenCV are right-handed. You have to probably flip the Z axis just as in the `viewer/unity_demo_stickers.py` [lines 130-132](https://github.com/jdibenes/hl2ss/blob/775469f8f1618f24b20a253fcea40afee849f5a0/viewer/unity_demo_stickers.py#L130).

Birkenpapier, 2023-03-20:
Thank you very, very much for your explanation!
The earliest I can test it is tommorrow morning so I'll come back to you again with my results.

Birkenpapier, 2023-03-21:
I've testet your suggestion and you were correct!
The rotation is now displayed correctly inside the unity application:
    Open3d + poinctloud registration result:

[image]

    Unity visualization on Hololens screenshot:
[image]

The last problem I need to solve now is the position/translation for the object(the head in this case) in unity to correctly place it after the registration.
Do you maybe have any idea what I can change to place it correctly inside unity's coordinate system? 
The red dot which you can see in the second image is the center of the application's coordinate system. 
My hand which is visualized as a green point cloud in the first image was in front of the red dot. The registration result is correct as you can see in the same image. After implementing your suggestions regarding the rotation the rotation is now correct inside the unity application leaving only the translation now which seems still to be mirrored.

I've tried to inverse it with the same method as the rotation and it works only when the object will gets recognized in front (0, 0, z) of the red dot otherwise it has offset to the hand in the real scene.

Thank you very, very much in advance again for your or any others help!

BR,

Birkenpapier