jdibenes/hl2ss #29: Would it be possible to run the appxbundle in the background on the hololens?

Michelvl92, 2023-03-29:
Currently, this is not possible. Is this possible, and what would be the best way to do this?

I want to show a Unity application in the hololens 2 (the Unity application is currently running on an external computer that is rendering this unity view)  and streamed to a viewer (that should be in the foreground). In the background, I want to run hl2ss server.

Birkenpapier, 2023-05-03:
May I ask you to elaborate your request? 
Do you want to render the data in the background and then displaying on another machine?

manzoned, 2023-10-26:
I would also be interested in this. I want to run the hl2ss app in the background to stream sensor data to python, send that data to unity via a socket, and then play the scene on the Hololens 2 via a Holographic Remote Player. For that to work both the hl2ss app and the holographic remote player would need to run simultaneously.

jdibenes, 2023-10-27:
It seems it is not possible to run two holographic applications simultaneously.
Here is an experimental 2D version of the appxbundle that can run alongside other apps:
https://github.com/jdibenes/hl2ss/blob/main/etc/hl2ss_1.0.26.0_arm64_FLAT.appxbundle

manzoned, 2023-10-27:
Thanks for sharing this. I can confirm that this 2D version can be run alongside other apps (e.g., Holographic Remoting Player).

However, I did notice some differences when running some of the sample codes in the viewer folder. For example, streaming the camera sensors works and plays, but the spatial input codes do not work properly/as with the original app bundle. The codes run, but the head, gaze and hand tracking do not provide proper values and are not projected onto the video frame.

zhangzhousuper, 2024-07-03:
Any updates on this issue? Also interested, thx