jdibenes/hl2ss #34: Incorrect timestamp when capturing more than one streams data

KellyYe, 2023-04-06:
Hi,

When capturing only one stream, like VLC left front, the timestamp is correct and the gap between timestamps is about 33ms. But when capturing more than one stream, even only two streams like VLC left front and right front, it will lose some frames and the gap between timestamp will become unstable like 33, 66, 33, 33, ...66, 99. 
Now my HL2 OS version is 10.0.19041.1467. Also, in newer OS version 10.0.20348.1440 the captured VLC frames will in wrong order. For example, the VLC frames which should be 1, 2, 3, 4, 5, 6  became 1, 2, 5, 6, 4 in timestamp increasing order. 
What's your OS version of HL2?

Thanks in advance!
Kelly

KellyYe, 2023-04-06:
[image]

KellyYe, 2023-04-06:
[image]

KellyYe, 2023-04-06:
[image]

KellyYe, 2023-04-06:
This is my code to capture data, it initializes two streams: left_front and right_front and only captures left_front timestamp for checking. 

`from pynput import keyboard

import numpy as np
import multiprocessing as mp
import cv2
import hl2ss_imshow
import hl2ss
import hl2ss_mp

# Settings --------------------------------------------------------------------

# HoloLens address
host = '169.254.13.130'

# Ports
ports = [
    hl2ss.StreamPort.RM_VLC_LEFTFRONT,
    hl2ss.StreamPort.RM_VLC_RIGHTFRONT
    ]

# RM VLC parameters
vlc_mode    = hl2ss.StreamMode.MODE_1
vlc_profile = hl2ss.VideoProfile.H264_BASE
vlc_bitrate = 1*1024*1024

# IMU: Maximum bytes to receive per step
# RM IMU parameters
imu_mode = hl2ss.StreamMode.MODE_1

# Maximum number of frames in buffer
buffer_elements = 60

#------------------------------------------------------------------------------

if __name__ == '__main__':
    client_rc = hl2ss.ipc_rc(host, hl2ss.IPCPort.REMOTE_CONFIGURATION)
    client_rc.open()

    producer = hl2ss_mp.producer()
    producer.configure_rm_vlc(True, host, hl2ss.StreamPort.RM_VLC_LEFTFRONT, hl2ss.ChunkSize.RM_VLC, vlc_mode, vlc_profile, vlc_bitrate)
    producer.configure_rm_vlc(True, host, hl2ss.StreamPort.RM_VLC_RIGHTFRONT, hl2ss.ChunkSize.RM_VLC, vlc_mode, vlc_profile, vlc_bitrate)
    
    for port in ports:
        producer.initialize(port, buffer_elements)
        producer.start(port)

    manager = mp.Manager()
    consumer = hl2ss_mp.consumer()
    sinks = {}

    for port in ports:
        sinks[port] = consumer.create_sink(producer, port, manager, None)
        sinks[port].get_attach_response()

    enable = True

    def on_press(key):
        global enable
        enable = key != keyboard.Key.esc
        return enable

    listener = keyboard.Listener(on_press=on_press)
    listener.start()

    file = open("gt.txt", "w")
    prevTime_lf = 0
    prevTime_rf = 0

    while (enable):
        for port in ports:
            _, data = sinks[port].get_most_recent_frame()
            if (data is not None):
                # all streams's pose are the same because they actually all get head_pose (but we only need one gt so that's okay)
                if (hl2ss.get_port_name(port) == 'rm_vlc_leftfront' and data.timestamp - prevTime_lf > 0):
                    #print(hl2ss.get_port_name(port) + ' Pose at time ' + str(data.timestamp))
                    #print(data.pose) 
                    #file.write('Pose at time ' + str(data.timestamp) + '\n')
                    #file.write(str(data.pose) + "\n")
                    #payload_cw_90 = np.rot90(data.payload, -1)
                    #cv2.imshow(hl2ss.get_port_name(port), payload_cw_90)
                    #cv2.imwrite(hl2ss.get_port_name(port)+"\\" + str(data.timestamp)+".jpg", payload_cw_90)
                    print(f"left front timestamp gap: {str(data.timestamp - prevTime_lf)}")
                    if (data.timestamp - prevTime_lf > 400000):
                        print("************************** lose left frame!")
                    prevTime_lf = data.timestamp

                '''elif (hl2ss.get_port_name(port) == 'rm_vlc_rightfront' and data.timestamp - prevTime_rf > 0):
                    payload_ccw_90 = np.rot90(data.payload, 1)
                    cv2.imwrite(hl2ss.get_port_name(port)+"\\" + str(data.timestamp)+".jpg", payload_ccw_90)
                    print(f"right front: {str(data.timestamp - prevTime_rf)}")
                    if (data.timestamp - prevTime_rf > 400000):
                        print("************************** lose right frame!")
                    prevTime_rf = data.timestamp'''

        #cv2.waitKey(1)

    for port in ports:
        sinks[port].detach()

    for port in ports:
        producer.stop(port)

    file.close()
    client_rc.close()
    listener.join()`

[image]

KellyYe, 2023-04-06:
This is the code to capture two streams data. It initializes VLC left_front and right_front but only capture left_front timestamps for checking. 

```
from pynput import keyboard

import numpy as np
import multiprocessing as mp
import cv2
import hl2ss_imshow
import hl2ss
import hl2ss_mp

# Settings --------------------------------------------------------------------

# HoloLens address
host = '169.254.13.130'

# Ports
ports = [
    hl2ss.StreamPort.RM_VLC_LEFTFRONT,
    hl2ss.StreamPort.RM_VLC_RIGHTFRONT
    ]

# RM VLC parameters
vlc_mode    = hl2ss.StreamMode.MODE_1
vlc_profile = hl2ss.VideoProfile.H264_BASE
vlc_bitrate = 1*1024*1024

# IMU: Maximum bytes to receive per step
# RM IMU parameters
imu_mode = hl2ss.StreamMode.MODE_1

# Maximum number of frames in buffer
buffer_elements = 60

#------------------------------------------------------------------------------

if __name__ == '__main__':
    client_rc = hl2ss.ipc_rc(host, hl2ss.IPCPort.REMOTE_CONFIGURATION)
    client_rc.open()

    producer = hl2ss_mp.producer()
    producer.configure_rm_vlc(True, host, hl2ss.StreamPort.RM_VLC_LEFTFRONT, hl2ss.ChunkSize.RM_VLC, vlc_mode, vlc_profile, vlc_bitrate)
    producer.configure_rm_vlc(True, host, hl2ss.StreamPort.RM_VLC_RIGHTFRONT, hl2ss.ChunkSize.RM_VLC, vlc_mode, vlc_profile, vlc_bitrate)
    
    for port in ports:
        producer.initialize(port, buffer_elements)
        producer.start(port)

    manager = mp.Manager()
    consumer = hl2ss_mp.consumer()
    sinks = {}

    for port in ports:
        sinks[port] = consumer.create_sink(producer, port, manager, None)
        sinks[port].get_attach_response()

    enable = True

    def on_press(key):
        global enable
        enable = key != keyboard.Key.esc
        return enable

    listener = keyboard.Listener(on_press=on_press)
    listener.start()

    file = open("gt.txt", "w")
    prevTime_lf = 0
    prevTime_rf = 0

    while (enable):
        for port in ports:
            _, data = sinks[port].get_most_recent_frame()
            if (data is not None):
                # all streams's pose are the same because they actually all get head_pose (but we only need one gt so that's okay)
                if (hl2ss.get_port_name(port) == 'rm_vlc_leftfront' and data.timestamp - prevTime_lf > 0):
                    #print(hl2ss.get_port_name(port) + ' Pose at time ' + str(data.timestamp))
                    #print(data.pose) 
                    #file.write('Pose at time ' + str(data.timestamp) + '\n')
                    #file.write(str(data.pose) + "\n")
                    #payload_cw_90 = np.rot90(data.payload, -1)
                    #cv2.imshow(hl2ss.get_port_name(port), payload_cw_90)
                    #cv2.imwrite(hl2ss.get_port_name(port)+"\\" + str(data.timestamp)+".jpg", payload_cw_90)
                    print(f"left front timestamp gap: {str(data.timestamp - prevTime_lf)}")
                    if (data.timestamp - prevTime_lf > 400000):
                        print("************************** lose left frame!")
                    prevTime_lf = data.timestamp

                '''elif (hl2ss.get_port_name(port) == 'rm_vlc_rightfront' and data.timestamp - prevTime_rf > 0):
                    payload_ccw_90 = np.rot90(data.payload, 1)
                    cv2.imwrite(hl2ss.get_port_name(port)+"\\" + str(data.timestamp)+".jpg", payload_ccw_90)
                    print(f"right front: {str(data.timestamp - prevTime_rf)}")
                    if (data.timestamp - prevTime_rf > 400000):
                        print("************************** lose right frame!")
                    prevTime_rf = data.timestamp'''

        #cv2.waitKey(1)

    for port in ports:
        sinks[port].detach()

    for port in ports:
        producer.stop(port)

    file.close()
    client_rc.close()
    listener.join()
```

[image]

jdibenes, 2023-04-06:
Hi,

>What's your OS version of HL2?

22621.1064 (windows insider program dev channel)

Please try the following script, get_most_recent_frame allows frame-skipping

```
from pynput import keyboard

import numpy as np
import multiprocessing as mp
import cv2
import hl2ss_imshow
import hl2ss
import hl2ss_mp

# Settings --------------------------------------------------------------------

# HoloLens address
host = '192.168.1.7'

# Ports
ports = [
    hl2ss.StreamPort.RM_VLC_LEFTFRONT,
    hl2ss.StreamPort.RM_VLC_RIGHTFRONT
    ]

# RM VLC parameters
vlc_mode    = hl2ss.StreamMode.MODE_1
vlc_profile = hl2ss.VideoProfile.H264_BASE
vlc_bitrate = 1*1024*1024

# IMU: Maximum bytes to receive per step
# RM IMU parameters
imu_mode = hl2ss.StreamMode.MODE_1

# Maximum number of frames in buffer
buffer_elements = 60

#------------------------------------------------------------------------------

if __name__ == '__main__':
    client_rc = hl2ss.ipc_rc(host, hl2ss.IPCPort.REMOTE_CONFIGURATION)
    client_rc.open()

    producer = hl2ss_mp.producer()
    producer.configure_rm_vlc(True, host, hl2ss.StreamPort.RM_VLC_LEFTFRONT, hl2ss.ChunkSize.RM_VLC, vlc_mode, vlc_profile, vlc_bitrate)
    producer.configure_rm_vlc(True, host, hl2ss.StreamPort.RM_VLC_RIGHTFRONT, hl2ss.ChunkSize.RM_VLC, vlc_mode, vlc_profile, vlc_bitrate)
    
    for port in ports:
        producer.initialize(port, buffer_elements)
        producer.start(port)

    manager = mp.Manager()
    consumer = hl2ss_mp.consumer()
    sinks = {}
    framestamps = {}

    for port in ports:
        sinks[port] = consumer.create_sink(producer, port, manager, None)
        framestamps[port] = sinks[port].get_attach_response() + 1

    enable = True

    def on_press(key):
        global enable
        enable = key != keyboard.Key.esc
        return enable

    listener = keyboard.Listener(on_press=on_press)
    listener.start()

    file = open("./data/gt.txt", "w")
    prevTime_lf = 0
    prevTime_rf = 0

    while (enable):
        for port in ports:
            state, data = sinks[port].get_buffered_frame(framestamps[port])
            if (state == 0):
                framestamps[port] += 1

                # all streams's pose are the same because they actually all get head_pose (but we only need one gt so that's okay)
                if (hl2ss.get_port_name(port) == 'rm_vlc_leftfront' and data.timestamp - prevTime_lf != 0):
                    #print(hl2ss.get_port_name(port) + ' Pose at time ' + str(data.timestamp))
                    #print(data.pose) 
                    #file.write('Pose at time ' + str(data.timestamp) + '\n')
                    #file.write(str(data.pose) + "\n")
                    #payload_cw_90 = np.rot90(data.payload, -1)
                    #cv2.imshow(hl2ss.get_port_name(port), payload_cw_90)
                    #cv2.imwrite(hl2ss.get_port_name(port)+"\\" + str(data.timestamp)+".jpg", payload_cw_90)
                    print(f"left front timestamp gap: {str(data.timestamp - prevTime_lf)}")
                    if (data.timestamp - prevTime_lf > 400000):
                        print("************************** lose left frame!")
                    if (data.timestamp - prevTime_lf < 0):
                        print("########################## ???????????????!")
                    prevTime_lf = data.timestamp

                '''elif (hl2ss.get_port_name(port) == 'rm_vlc_rightfront' and data.timestamp - prevTime_rf > 0):
                    payload_ccw_90 = np.rot90(data.payload, 1)
                    cv2.imwrite(hl2ss.get_port_name(port)+"\\" + str(data.timestamp)+".jpg", payload_ccw_90)
                    print(f"right front: {str(data.timestamp - prevTime_rf)}")
                    if (data.timestamp - prevTime_rf > 400000):
                        print("************************** lose right frame!")
                    prevTime_rf = data.timestamp'''

        #cv2.waitKey(1)

    for port in ports:
        sinks[port].detach()

    for port in ports:
        producer.stop(port)

    file.close()
    client_rc.close()
    listener.join()
```

KellyYe, 2023-04-15:
Thanks for your reply! Now HL2 can capture vlc streams without losing frames. I really appreciate your help.

closed 2023-04-15 by KellyYe