jdibenes/hl2ss #47: Is cameraToWorld matrix involving a convert from right hand to left hand ?

kookoo9999, 2023-07-13:
Thank you for your sharing a nice code!

In order to align virtual models made with 3D slicers (right-handed coordinate), I have to do several coordinate system transformations (3D Slicer(right-handed)->Optical Tracking System(right-handed)->Holens2 Depth Camera(maybe right-handed)->Unity(left-hanede)).
At this time, I lack an understanding of the camera_to_world matrix that converts to Depth Camera->Unity.

First of all, In unity_sample_stickers.py script,  I found 4x4 matrix called 'camera to world' . 
I think it's transform matrix converting from depth camera coordinate to Unity world coordinate.
If I use a point(x,y,z) of depth camera coordinate , camera_to_world * [x,y,z,1].T and have to flip z-axis (z -> -z).
1. Is camera_to_world matrix  just right handed coordinate(depth camera coordinate) -> right handed coordinate(z-fliped Unity coordinate) ?? 
2. Why do I flip z-axis after multiply camera_to_world matrix by point of depth coordinate ??
3. When I try to use the rotation of the camera_to_world matrix, do I not need to convert right-handed coordinate to left-handed coordinate??

Thank you for your reading!
I hope you have a nice day!

jdibenes, 2023-07-13:
Hi.

1) Yes. The `camera2world` matrix in `unity_sample_stickers` converts from depth camera coordinates to world coordinates (both right handed). So for a 3D point `[x, y, z]` in depth camera coordinates we have `world_point_homogeneous = [x, y, z, 1] * camera2world`.
2) Negating the z component of the world points converts them to Unity coordinates (left handed). IIRC for hololens world right=+x, up=+y, forward=-z and for Unity right=+x, up=+y, forward=+z. 
3) Not in this case since rotation is already considered when applying camera2world to the 3D points.

See [Converting between coordinate systems](https://learn.microsoft.com/en-us/windows/mixed-reality/develop/unity/unity-xrdevice-advanced?tabs=mrtk) for the formulas used to convert 3D points, rotations (as quaternions), and 4x4 matrices between left and right coordinates.

kookoo9999, 2023-07-17:
Thank you very very much @jdibenes !
I'm late to check your comment.
'world' means IIRC for hololens in camera2world. I'll try it again :)

jdibenes, 2023-07-22:
I'm not familiar with 3D Slicer and Optical Tracking System but maybe even though they are right-handed the orientations of their coordinate systems and the HoloLens world are different. For example, for some CAD software Z is up instead of Y. If this is the case, then you would have to apply additional rotations to align all the coordinate systems.
I think determinant -1 is expected since right to left hand conversion involves a reflection.

kookoo9999, 2023-07-27:
@jdibenes 
Thank you for your answer!
Actually We're on same page. because there's a different of coordinate between optical tracking system(OTS) and Hololens.
            OTS          IIRC for hololens world
right :  +x        ,                +x
up :     -y         ,                +y
front : +z         ,                -z

I think since last converting coordinate is OTS so right convert is filp y-axis.
So I use result quaternion as x,-y,-z,-w but it doesn't work..

kookoo9999, 2023-08-07:
@jdibenes 
Thank you very very much for your answer!
I tryna test using pv camera by sample_intergrator_pv.py.
Actually I used this code to get camera2world(pv to world). because in other script, I found that depth camera to world is similary below code.
Is it correct?

pv2world = hl2ss_3dcv.camera_to_rignode(color_extrinsics) @ hl2ss_3dcv.reference_to_world(data_pv.pose)

kookoo9999, 2023-10-10:
@jdibenes 
Hello , thank you for sharing your knowledge .

I have a question.

First , you can see the figure below(long throw mode). I don't know why there are no datas in center of ball.
[image]

Second , when I use depth sensor as AHAT , is there same way as I get a matrix to convert from depth to world(camera2world) in long throw mode??

Third, can you tell me where direction of PV camera is  ? (e.g front : z, up : -y , right x )

thank you for reading.

I hope you have a nice day :)

jdibenes, 2023-10-12:
Hello,

>First , you can see the figure below(long throw mode). I don't know why there are no datas in center of ball. I use reflective marker like below figure.

The depth camera has issues with dark or shiny objects, so maybe that's the cause.

>Second , when I use depth sensor as AHAT , is there same way as I get a matrix to convert from depth to world(camera2world) in long throw mode??

Yes, it's the same formula `ahat_to_world = hl2ss_3dcv.camera_to_rignode(calibration_ht.extrinsics) @ hl2ss_3dcv.reference_to_world(data_ht.pose)`.

>Third, can you tell me where direction of PV camera is ? (e.g front : z, up : -y , right x )

I think it's front: -z, up: y, right: x.
See `sample_integrator_pv.py` and `hl2ss_3dcv.pv_fix_calibration`. Original frame front: z, up: -y, right: x is rotated 180 deg about the x axis.

kookoo9999, 2023-10-13:
@jdibenes 
Thank you for your answer! :)

closed 2023-10-13 by kookoo9999