jdibenes/hl2ss #51: Return X, Y, Z coordinates

vstadnytskyi-FDA, 2023-07-18:
Is it possible to return a stream of X,Y,Z coordinates of where HoloLens2 thinks it is?

jdibenes, 2023-07-19:
The Spatial Input stream captures head position and orientation @ 30 Hz.
See `head_pose.position` in [client_si.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/client_si.py).

vstadnytskyi-FDA, 2023-07-19:
And also it is very nicely explained in the paper - https://arxiv.org/pdf/2211.02648.pdf

[image]

If I understand it correctly, we get X,Y,Z coordinates. Why do we need to more vectors - forward and up to define the direction in which head is pointing.

jdibenes, 2023-07-22:
Those vectors encode the orientation of the head.