jdibenes/hl2ss #52: Get ray from coordinate in PV camera frame

Zzx-J, 2023-07-20:
Hi, Really thanks for your great work, it helps a lot!

I just wonder is it possible to get the coordinate of a point in the world space from the coordinate in the pixel space of the PV camera frame with the help of sm_manager?

I noticed that for hand and eye tracking in sample_si_pv.py, the gaze point in the world space is acquired through these code below:

            eye_ray = si.get_eye_ray()
            eye_ray_vector = hl2ss_utilities.si_ray_to_vector(eye_ray.origin, eye_ray.direction)
            d = sm_manager.cast_rays(eye_ray_vector)
            if (np.isfinite(d)):
                gaze_point = hl2ss_utilities.si_ray_to_point(eye_ray_vector, d)

where we get eye ray vector and get the distance to the gaze point by casting the eye ray using sm_manager, then we can get the gaze point in the world space using "hl2ss_utilities.si_ray_to_point"

So I suppose If we can get a ray vector from the pixel coordinate in the PV camera frame and get the distance using sm_manager.cast_rays, then we can have the object coordinate in the world space.

I know we can get the coordinates of a point in the world space using pixel coordinates from Depth camera, but the PV camera frame is necessary for me and there is a known issue for getting PV and depth frame simultaneously and unfortunately I can't solve that.

jdibenes, 2023-07-20:
Hi,
I think it could work.
1) Compute the origin of the PV camera `pv_origin_h = [0,0,0,1] * pv_pose`.
2) Compute the ray corresponding to the PV pixel [u, v] `pixel_direction = unit([u, v, 1] * inv(pv_intrinsics[:3, :3]) * pv_pose[:3, :3])`.
3) Use origin and direction to compute `d` via raycasting.
4) The 3D point (in world coordinates) for the pixel [u,v] is then `pv_origin + d*pixel_direction`.

Zzx-J, 2023-07-20:
Very clear, I will try it tomorrow, thanks for your reply!

jdibenes, 2023-07-22:
We added an example based on this:
[sample_sm_pv_depth.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_sm_pv_depth.py)

NTUZZH, 2023-08-15:
I just wonder if it is possible to get the 3D coordinate of a point in the world space from the 2D coordinate in the pixel space of one Depth image under long_throw mode?

closed 2024-01-30 by Zzx-J