jdibenes/hl2ss #53: Converting pixels to 3D world coordinate system

gonalo99, 2023-07-20:
Hi, I'm incredibly grateful for your outstanding work,it has been incredibly helpful!

What would be the process to get the 3D location in world coordinate system of a pixel that we get in the PV image?

Currently, I'm employing this process:
  1. inverting the projection, using the PV intrinsics, to obtain the ray where the pixel lies. 
  2. Using the closest aligned depth frame, I get the depth value of that pixel
  3. With the ray and the depth, I can get the x,y,z coordinates in the camera reference system
  4. This is the part I have most doubts about. Currently, I'm directly applying the pose that comes coupled with the PV frame to transform the camera reference system coordinate to the world coordinate system, using the `reference_to_world` function and homogenous coordinates.

However, when I draw the resulting point in Unity's coordinate system it doesn't look correct. Am I right to assume that the PV pose is the transformation from PV_camera to world? Or do I need to apply another transformation before?

jdibenes, 2023-07-22:
Hi,

Currently, there is no functionality in the library to do depth-to-PV alignment.

How are you aligning the PV and depth frames?
If you are using the same procedure as the samples (PV-to-depth) then the rays corresponding to the pixels of the aligned PV image are actually in depth camera coordinates so you have to use the depth intrinsics/extrinsics/pose.
Point 4 is correct: if you have a 3D point in PV coordinates then the pose of the PV frame converts to world coordinates (using `reference_to_world` as you mentioned). For converting PV pixels to rays in PV coordinates you have to use `hl2ss_3dcv.pv_fix_calibration` and use the returned intrinsics/extrinsics along with the pose, see [sample_sm_pv_depth.py](https://github.com/jdibenes/hl2ss/blob/eee7c6ef0ea480affb0f08d459d08cca306ef49b/viewer/sample_integrator_pv.py#L129C57-L129C75).
Are you converting the 3D points to left-handed coordinates when passing the points to Unity?

As for alternatives, there is the trick proposed in #21.
And it is possible to generate depth maps for PV frames using the Spatial Map in the HoloLens but it seems the map updates slowly so it may be suitable only for static scenes [sample_sm_pv_depth.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_sm_pv_depth.py).

gonalo99, 2023-07-22:
I'm using the pixels in the PV coordinates, and I get the corresponding depth in a similar way to the trick in https://github.com/jdibenes/hl2ss/issues/21. Specifically, after having the maps from RGBD registration, I look for depth values in a small circle around the corresponding PV pixel (since there might not be a perfect correspondence) and just take the minimum depth inside that circle.

With this infomation, I am calculating the 3D coordinates in camera coordinate frame using the intrinsics information (Fx, Fy, Cx and Cy):
  1. X_3D = Z * (X_pixel - Cx) / Fx
  2. Y_3D = Z * (Y_pixel - Cy) / Fy
  3. Z_3D = Z

So according to `hl2ss_3dcv.pv_fix_calibration`, I have to reverse the sign of the Fx (intrisics[0,0] value in the intrinsics matrix before using it to calcule the 3D coordinates of the point, right? I wasn't doing that, so it might be the error in my calculations.

I'm not sure if I'm converting the points properly when passing them to Unity. Can you provide me more details about the expected 3D coordinate system in the HoloLens, assuming the 3D coordinates are properly calculated?

jdibenes, 2023-08-02:
1. Yes. Also use the new extrinsics.
2. You're right, use pose -> extrinsics -> intrinsics.
3. For 3D points in HoloLens world coordinates (right handed), flip the sign of the Z component to convert to Unity coordinates (left handed). See [Converting between coordinate systems](https://learn.microsoft.com/en-us/windows/mixed-reality/develop/unity/unity-xrdevice-advanced?tabs=wsa).

Also, I think you can just compute the 3D point in world coordinates using the depth camera intrinsics/extrinsics/pose (corresponding to the minimum depth in the circle), and that 3D point corresponds to the PV pixel. If you need the actual depth value then compute norm(point - pv_origin), where pv_origin is the position of the PV camera in world coordinates (last row of pose).

gonalo99, 2023-08-02:
Thank you so much, that was very helpful.