jdibenes/hl2ss #58: Cannot get camera frame with unity sample project

fmtibaa, 2023-08-22:
Dear author,

I'm testing your unity sample application on HoloLens 2 but i can not get data from the RGB and depth camera for point cloud reconstruction (when i launch the python script to get camera frames i get nothing displayed on my desktop). Is there any manipulation in the player setting or in the HoloLens to allow getting data from camera sensors ?

I'm using unity 2020.3.47f1.

Thank you in advance for your answer.

jdibenes, 2023-08-22:
Hello,

What script are you running?
Is your desktop OS Linux?
Does the Unity sample have access to the camera? (Settings -> Privacy -> Camera, hl2ss sample should be enabled)
Can you try running sample_integrator_pv.py?

jdibenes, 2023-08-23:
[sample_integrator_pv.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_integrator_pv.py)

fmtibaa, 2023-08-24:
Hello,

Yes i tried it and it works fine. Thank you, good work !!!

closed 2023-08-24 by fmtibaa