jdibenes/hl2ss #59: python client can't obtain video from Hololens2.

Xbathy, 2023-08-24:
[image]
 When start python , will exit with error.
python 3.10
hl2ss    1.0.14

jdibenes, 2023-08-24:
Hello,
The current python scripts are not compatible with older versions of hl2ss.
Does the problem still happen if you use hl2ss 1.0.23?

Xbathy, 2023-08-25:
When i use hl2ss 1.0.23, app will crash.

https://github.com/jdibenes/hl2ss/assets/63304916/b54f31fa-c3bb-4725-88bb-e63d40ce9e0f

hololens2 system : 20348.1432

jdibenes, 2023-08-25:
Please check that hl2ss.exe has access to the camera (Settings -> Privacy -> Camera), eye tracker (Settings -> Privacy -> Eye tracker), microphone (Settings -> Privacy -> Microphone), and user movements (Settings -> Privacy -> User movements).
Also check that the HoloLens is connected to the network before launching the application (it will crash otherwise).
If that does not work, try updating your HoloLens.

Xbathy, 2023-08-25:
Good, I have update Hololens2 to 22621.1125, works fine. And looks like can obtain front camera stream and depth camera stream simultaneously now.
More feedback will be given after more testing.

closed 2023-08-25 by Xbathy