jdibenes/hl2ss #65: How load obj file during running

Civs17, 2023-09-12:
Hi, I'm trying to load a obj file, saved in a server, while the app unity is running on the hololens. I've created a new command to send the url of the file and download it with the same method of OBJLoader project. But It seems to not work. Someone has an idea? 

public uint MSG_LoadOBJFile(byte[] data)
    {
        //string url = Encoding.UTF8.GetString(data);
        // Create a UnityWebRequest to download the OBJ file
        UnityWebRequest request = UnityWebRequest.Get("http://localhost:8081/spina_obj.obj");

        // Send the request and wait for it to complete
        request.SendWebRequest();
        while (!request.isDone)
        {
            System.Threading.Thread.Sleep(1);
        }

        // Check for errors during the download
        if (request.result != UnityWebRequest.Result.Success)
        {
            Debug.LogError("Failed to download OBJ file. Error: " + request.error);

        }

        // Get the downloaded data as bytes
        byte[] downloadedData = request.downloadHandler.data;

        // Create a stream and load the OBJ file
        var textStream = new MemoryStream(downloadedData);
        var loadedObj = new OBJLoader().Load(textStream);

        // Add the loaded object to your game
        uint id = AddGameObject(loadedObj);
        return id;
    }

jdibenes, 2023-09-13:
Hello,
Does the python code receive the id of the object (from `ipc_umq.pull`)?

Civs17, 2023-09-14:
Yes, this Is the idea.

bxcharlie, 2023-10-31:
Hello，Does the current code work? I also want to know how to load obj file, can you share your experience?