jdibenes/hl2ss #70: How do I convert data.timestamp to UTC time

chinaenergetic, 2023-11-08:
In the client stream series file I see a data.timestamp print. But I try to read and analyze the author's code and instructions. Due to my poor programming background, I couldn't figure out how to convert data.timestamp to UTC time. Can the author explain how to convert a timestamp to UTC via python? Thank you very much！

jdibenes, 2023-11-09:
First, obtain the UTC offset from the server:
```
client = hl2ss_lnm.ipc_rc(host, hl2ss.IPCPort.REMOTE_CONFIGURATION)
client.open()
utc_offset = client.get_utc_offset(32)
client.close()
```
Then add the UTC offset to the timestamps to convert them to UTC:
```
utc_timestamp = data.timestamp + utc_offset
```

javiercmh, 2024-01-29:
Dear developers, I'm having issues with the timestamps, even after following this solution. I am assuming that all timestamps are in nanoseconds.

```
data.timestamp: 146716183355 (1970-01-01 01:02:26.716183)
utc_offset: 133509951844495077 (1974-03-26 07:05:51.844495)
data.timestamp + utc_offset: 133510098560678432 (1974-03-26 07:08:18.560678)
current_time: 1.7065362561664888e+18 (2024-01-29 14:50:56.166489)
current_time - utc_offset: 1.5730263043219937e+18 (2019-11-06 08:45:04.321994)
current_time - (data.timestamp + utc_offset): 1.5730261576058104e+18 (2019-11-06 08:42:37.60581)
```

It seems like the timestamp is not related to the UNIX epoch (1970-1-1) but to some other reference point. Any suggestions? What am I doing wrong?

Thanks in advance!

jdibenes, 2024-01-29:
Hello,
All timestamps are in hundreds of nanoseconds (1 timestamp unit = 100 ns).
The `utc_offset` given by the server converts timestamps to [Windows Filetime](https://learn.microsoft.com/en-us/windows/win32/sysinfo/file-times).