jdibenes/hl2ss #72: Error: invalid vs_2_0 output semantic 'SV-TARGET'

JVal123, 2023-11-17:
I was looking to install the hl2ss app in the Hololens-2, however whenever I run the code on VS Studio 2019 this error appears and I can´t quite figure out what it stands for and why it happens.

MayooranT, 2023-11-20:
I think you can try this.
Open the Property page for the .hlsl file, and in HLSL Compiler/General/Shader Type select Pixel Shader.

jdibenes, 2023-12-06:
Hello,
This project requires the v143 Platform Toolset which (afaik) is for Visual Studio 2022.
Try building using Visual Studio 2022 and see if the errors persist.

JVal123, 2023-12-11:
Hello again! After trying to build with Visual Studio 2022, the errors still persist. The images below show the errors I'm talking about.

Here, we can see that v143 Platform Toolset is selected:
[image]

After running the remote machine with the ARM64 architecture and the following shader settings:
[image]

we get this error:

[image]

If I select the pixed shader: 
[image]

we get another of the already above mentioned errors:

[image]

Also, in case I change the shader model to 5.1, to try and solve error X4016, several X4502 errors pop up. This last error also appears if I use this shader model, but without selecting a shader type.

jdibenes, 2023-12-11:
Your images show that you're building as Debug ARM64, which is not supported.
Try building as Release ARM64.

closed 2024-01-05 by JVal123

vli-syngular, 2025-02-13:
> Your images show that you're building as Debug ARM64, which is not supported. Try building as Release ARM64.

Do you mean we don't have the permission to build as Debug? So how can I debug the codes?

jdibenes, 2025-02-13:
No, it's just that the debug profile was never configured. 
If you want to build as Debug ARM64 just set the C++/Linker settings like the Release profile (except those pertaining to debug/optimizations of course).