jdibenes/hl2ss #75: How can I get the pv2depth transform matrix?

kookoo9999, 2023-12-07:
Thank you for sharing your nice code.

I wanna get the transform matrix(pv camera -> depth sensor(long throw).

1.Can you tell me how I can get the matrix?

And As I know depth sensor(long throw) is right handed coordinate but I think it's left handed coordinate because I got the 4 points and find it's direction up:-y right: -x front:+z.

2.Can you tell me where direction of PV and depth sensor(longh throw) is?

Thank you for reading!

I hope you have a nice day :)

jdibenes, 2023-12-09:
Hello,
See [sample_rgbd_pv.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_rgbd_pv.py), lines 115-120, where the long throw to pv camera matrix is computed and is used to transform 3D points from long throw to pv.
The pv camera to long throw transform is the inverse of this matrix.
All coordinate systems are right-handed up:y, forward:-z, right:x.

kookoo9999, 2023-12-13:
Thank you @jdibenes .

I tried to test again and I found depth sensor coordinate use a right handed.

Your answer was very helpful!

closed 2023-12-13 by kookoo9999