jdibenes/hl2ss #77: Increased Latency in PV Video Streaming When Deployed as Unity Plugin

xli149, 2023-12-11:
Hello, thank you very much for being able to provide us with such a great open-source project. After installing the appxbundle on Hololens 2, I can successfully access PV video streaming, and the latency is relatively low. However, when I deploy it as a Unity plugin, I find that the latency of PV video streaming increases significantly, to about 1 second. Why is this happening? Is there any way to reduce latency?

Best Regards

jdibenes, 2023-12-11:
Hello,
We believe this happens because running Unity + 1080p video compression is too demanding for the HoloLens hardware.
A possible workaround is to reduce PV resolution and/or framerate.

xli149, 2023-12-13:
> Hello, We believe this happens because running Unity + 1080p video compression is too demanding for the HoloLens hardware. A possible workaround is to reduce PV resolution and/or framerate.

Thank you very much for your reply. If it's convenient, could you please let me know specifically which part of the code should be modified? I haven't found the location to adjust the resolution yet.

Best Regards

jdibenes, 2023-12-14:
See [client_stream_pv.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/client_stream_pv.py) lines 32-35 and 79.
Supported resolutions and framerates are in [pv_configurations.txt](https://github.com/jdibenes/hl2ss/blob/main/etc/pv_configurations.txt).