jdibenes/hl2ss #82: how to use hl2ss_mt.cpp in C++ client library

lishiyu005, 2024-01-09:
Dear author,

I meet some problems using the hl2ss_mt.cpp file in C++ client library. Is there any code to show how to use multithreading of the Hololens stream?

Here is my code which does not work.

#include "hl2ss_mt.h"
#include "hl2ss_lnm.h"
#include <opencv2/opencv.hpp>

// Initialize personal video of HOLOLENS
char const* host = "10.184.2.207";
uint16_t port = hl2ss::stream_port::PERSONAL_VIDEO;
std::unique_ptr<hl2ss::rx_pv> client = hl2ss::lnm::rx_pv(host, port, width, height, framerate);
std::string port_name = hl2ss::get_port_name(port);

hl2ss::lnm::start_subsystem_pv(host, port, enable_mrc);

std::cout << "Downloading calibration for " << port_name << " ..." << std::endl;
std::shared_ptr<hl2ss::calibration_pv> calibration = hl2ss::download_calibration_pv(host, port, width, height, framerate);
std::cout << "Done." << std::endl;

// Start multithreading
auto source_ptr =
      std::make_shared<hl2ss::mt::source>(300, std::move(client));
 source_ptr->start();

// Get image from buffer and save
  int64_t frame_index = 1;
  int32_t status = 0;
while (true){
  data = source_ptr->get_packet(frame_index, status);
  frame_index++;
  uint8_t *image;
  hl2ss::pv_intrinsics *intrinsics;
  hl2ss::unpack_pv(data->payload.get(), data_color_->sz_payload, &image, &intrinsics);
  cv::Mat image_ = cv::Mat(intrinsics_.height, intrinsics_.width, CV_8UC3, image);
  cv::imwrite("test.png", image_);
}

jdibenes, 2024-01-10:
Hello,
We have added an example in [/extensions/client_cpp/main.cpp](https://github.com/jdibenes/hl2ss/blob/60c5177f90358920db02bcb6a24ac330c39d29de/extensions/client_cpp/main.cpp#L468).

lishiyu005, 2024-01-18:
Thank you very much. It works very well !

closed 2024-01-18 by lishiyu005