jdibenes/hl2ss #84: Server not running

theo-wner, 2024-01-14:
Hey there, I'm having problems running newer versions than 1.0.20.0.
Whenever i try accessing a stream, the connection to the server can't be established.
I guess the server isn't running at all.
When i start the server on my Hololens, the hl2ss task is running for a couple of seconds and then disappears.
Is this  somehow related to the Update "application now exits on suspend"?
Am i not starting the server properly?

jdibenes, 2024-01-15:
Hello,
What is your HoloLens OS version?

theo-wner, 2024-01-15:
Thanks for the quick reply! 
I'm currently using build 10.0.19041.1395

jdibenes, 2024-01-15:
Try updating your HoloLens.
The app does not work on 20348.1432 or older.

theo-wner, 2024-01-16:
Thanks a lot for your quick tips! My Hololens had some issues updating, which i could solve using Microsofts Advanced Recovery Companion for anyone dealing with the same issue.

closed 2024-01-16 by theo-wner