jdibenes/hl2ss #88: Depth Upsamling

AhmedAyman1, 2024-02-23:
First of all, thank you for your great work! 

I had a question regarding your pipeline, is it possible to upsamle the depth in order make better use of the high resolution PV camera with AHAT ?

jdibenes, 2024-02-25:
Hello,
It is possible, see [sample_pv_depth_ahat.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/sample_pv_depth_ahat.py).

jdibenes, 2024-02-29:
Use `color_intrinsics` to generate the rays. The last element for all rays should be 1.
Multiply rays by depth (`pv_z` in the script) to get the 3D points.
Transform the 3D points to world space using:
```
hl2ss_3dcv.camera_to_rignode(color_extrinsics) @ hl2ss_3dcv.reference_to_world(data_pv.pose)
```

AhmedAyman1, 2024-02-29:
I did exactly the same, including one extra step, which is incorrect. I was dividing the depth by the max_depth. After I removed the division, the issue was resolved.

Thank you

closed 2024-02-29 by AhmedAyman1