jdibenes/hl2ss #89: World reference frame is overwritten and placed almost 2 meters down

Alessandro98-Albanesi, 2024-02-27:
Good evening. The plugin is great and i have been using it for months now. I just keep having this problem. When i start the app in the hololens the world reference frame is overwritten and shifted down (it appears almost under the floor). I really can't get my mind around it. Is there any explanation why this happens_ Is there a way to avoid this and keep the original unity world reference frame in Hololens?
Thanks in advance for the answer.

Alessandro.

jdibenes, 2024-02-28:
Hello,
What Unity version are you using?
Are you using MRTK 2, MRTK 3, or something else?
Does this happen on the Unity sample provided in this repo?

Alessandro98-Albanesi, 2024-02-28:
Hi! Thanks for the prompt answer. I am using Unity 2021.3.31 and MRTK2. i have imported the plugin in my project as described in the repo and everything works fine, so the stream works but i just have this shift of the world coordinates. looking at the scripts i have noticed that in the hl2ss script in the unity project, there is this function 
_[DllImport("hl2ss")]
private static extern int OverrideWorldCoordinateSystem(IntPtr scs);_
what does this do exactly? Could it be related to the issue i am facing? Thanks

jdibenes, 2024-02-29:
It makes the plugin use the same coordinate system as the Unity scene to fix https://github.com/jdibenes/hl2ss/issues/27.
Try commenting out line 41 in the Hololens2SensorStreaming.cs script
 https://github.com/jdibenes/hl2ss/blob/17a56950dd0c931f3ccd966ca9fd7b8232add336/unity/Assets/Scripts/Hololens2SensorStreaming.cs#L41
Also what Alignment Type (in the MixedRealitySceneContent object) are you using and what Target Experience Scale (MixedRealityToolkit object)?
And how are you visualizing the world reference frame?
I'm trying to reproduce the issue.

Alessandro98-Albanesi, 2024-02-29:
Hello, i have already tried to comment the line you are referring to but nothing seems to change. In my project i am not including the Mixed Reality Scene Content. Shouldn't it be the same? Eventually the target experience scale is Room. What do you mean "How are you visualizing the world reference frame?".