jdibenes/hl2ss #90: AHAT-PV and cloud points

NguyenRichard, 2024-03-04:
Hi, 

First of all thank you very this awesome project as we have been using your code in our project for a while and cited your work.

As we wanted to explore AHAT-PV data, we have lightly modified your lt_pv sample to make it work for ahat.
We have also basically reversed the lt_pv matrix to create a parsed AHAT scaled up to the RGB resolution. As a result we have stored those data:
Raw AHAT 
(saved from:
        depth = hl2ss_3dcv.rm_depth_undistort(data.payload.depth, self.calibration.undistort_map)
        depth = hl2ss_3dcv.rm_depth_normalize(depth, self.scale)
 )
[image]
Scaled to PV AHAT
[image]
Cloud points
(from loading the stored depth and xy1, xy1 has been stored once when the hl2ss recording starts and is used for all frames until the recording ends.
then using:
    ht_points = hl2ss_3dcv.rm_depth_to_points(xy1, depth)
    cmap = plt.cm.plasma
    ax.scatter(ht_points[:, :, 0], -ht_points[:, :, 1], -ht_points[:, :, 2], c=ht_points[:, :, 2] / 5, cmap=cmap, s=1))
[image]

1) Main question: 
The cloud points seems to be very weird for the points "closer to the camera" (in purple) that I assume are supposed to represent the hand. The cloud points plot is simply the plot of the ahat_points (after hl2ss_3dcv.rm_depth_to_points()) with x right, y down and z forward. Do you have any idea where the problem could be coming from?
Notes: 
-I assume the data is fine as the fact that the Scaled to PV AHAT is matching the RGB image should means that the projections to world and then to pv should be correct. 
-Some of the objects in the RAW AHAT can be recognized in the plot of the cloud points while the hand is mostly the weird part.
-The points that are far away (ex: walls) are not shown in the point cloud plots either even if we plot the result from rm_depth_to_points without filtering anything.

2) We have tried to fill our Scaled to PV AHAT using some simple interpolation but they are very long to process. As such, your new AHAT_pv sample is very interesting as you have a real time upscaling of the reprojected AHAT. Is it possible for you to give some explanation on why you decided in the sample to split xy1_o and xy_d and how it helped you create the upscaled AHAT?

Best regards,
Richard

jdibenes, 2024-03-07:
Hello,

Q1) It's difficult to see, but it looks like those points close to the camera come from the floor, as they appear darker than the desk in the depth image (AHAT depth wraps around about each meter).

Q2) For each depth pixel d[y,x] we use its corresponding ray xy1[y,x,:] (xy1_o) and the bottom right ray xy1[y+1,x+1,:] (xy1_d), then we project both 3D points (d[y,x]*xy1[y,x,:] and d[y,x]*xy1[y+1,x+1,:]) to the PV image. Then, these two projections give the top left corner and the bottom right corner of a rectangular "superpixel" in the PV-depth image, and we fill all of the pixels inside that rectangle with the same depth (obtained from d[y,x]*xy1[y,x,:] after transforming to PV space).

NguyenRichard, 2024-03-07:
Hello,

Thanks for the answers!

QC1) Indeed, now that you mention it, the floor is  considered as close to the camera for some weird reason. I would have expected that when objects are farther than the range of hand support (~1m according to HoloLens2ForCV), they would have the 0 value. I will continue to explore how to use those data. 

QC2) Thanks for the explanation, it's more clear now and very ingenious! I will update our project to build directly the interpolated depth using our method.

closed 2024-03-07 by NguyenRichard