jdibenes/hl2ss #93: Difference sample-based & client-based streaming. Simultaneous AHAT & PV

anjelomerte, 2024-03-13:
In `simple_recorder.py` it say that "Simultaneous PV and RM Depth AHAT streaming is not supported [...]". Is that true? If so, why is that? It is definitely possible to access both simulatneously on the HL2.

Combining `client_stream_pv.py` and e.g. `sample_si_rm_depth_ahat.py` does work for a couple of seconds, then it freezes. Couldn't figure out why though. 
That also got me wondering what the different in streaming approach is between the `smaple[...].py` and `client[...].py` implementations?

jdibenes, 2024-03-14:
Hello,
The comments are deprecated, PV+AHAT did not work in some older OS versions (https://github.com/microsoft/HoloLens2ForCV/issues/133).
For the `sample[...].py` scripts that use multiple streams, the main difference is that we use multiprocessing (producer, consumer in hl2ss_mp) to get around the global interpreter lock.
When the scripts froze did they throw any errors?
Both `client_stream_pv.py` and `sample_si_rm_depth_ahat.py` use a lot of network bandwidth so it's possible they stopped working due to network conditions.

anjelomerte, 2024-03-15:
Hi, thanks for the insights! Makes sense now. No, I got no errors, the streams just froze.

closed 2024-03-15 by anjelomerte