jdibenes/hl2ss #97: Sending messages from server (HL2) to client (PC)

anjelomerte, 2024-03-27:
Hey, yet another question:) Thanks for being so active on your issues!

Is there a way to send messages from the server to the client? I see that e.g. when remotely creating scenes in Unity from the client, a result array is "pulled", so there seems to be some communication in this direction implemented.

In particular, I would like to start and stop the streaming from the server side. For that, the transmission of a simple timestamp would be sufficient. Is this possible without modification of the underlying C++ plugin?

jdibenes, 2024-03-29:
Hi,
There's no support for sending messages from HL2 to the PC in the same format as they are sent from PC to HL2.
You can try using the HL2->PC 4-byte message queue by itself but maybe it would be simpler to create a new command to poll the server every few milliseconds and determine whether to stop/start the stream or do nothing.
In either case, no modification of the C++ plugin is necessary, just need to change the RemoteUnityScene C# script and hl2ss_rus.py.

anjelomerte, 2024-04-26:
Hey jdibenes,

thanks for the reply and sorry for my late reply, sth got inbetween. Anyways, your suggestion with polling the server seemed very reasonable which is what I'm currently digging into. I have managed to implement some custom commands for now to test behavior. The client can send some stuff (e.g. text) to the server (confirmed). However, I am struggling to read the response e.g. in the follwing setting:

    # Create command buffer and append command(s)
    buffer = command_buffer() 
    buffer.tell_connect_success()

     # Send command(s) in buffer to the Unity app and receive response (4 byte unsigned integer per command)
    client.push(buffer)
    response = client.pull(buffer)

    # Evaluate response
    print(response[0]
    
In line `response = client.pull(buffer)` the program simply freezes. Why could that be? How can I read the uint response by the server? I need to be able to read the response to e.g. pack a timestamp as an integer. Thanks a lot!

jdibenes, 2024-04-29:
That code looks OK.
How are you implementing the `tell_connect_success` method?

anjelomerte, 2024-04-30:
I tried to follow your code examples. In the module we would have a class `command_buffer(hl2ss.umq_command_buffer)` like so:

```
# Custom command buffer -------------------------------------------------------
class command_buffer(hl2ss.umq_command_buffer):
    # Command structure
    # id:     u32 (4 bytes)
    # size:   u32 (4 bytes)
    # params: size bytes

    # Indicate successful connection
    def connect_success(self):
        # Command id: 0xFFFFFFFD
        # Command params: empty byte data (id enough to indicate purpose)
        self.add(0xFFFFFFFD, b"") 
```

Then we define some methods to call like so:

```
# Message functions -----------------------------------------------------------
# Inform server about succuessful connect
def inform_connect():
    # Create command buffer and append command(s)
    buffer = command_buffer() 
    buffer.connect_success()

     # Send command(s) in buffer to the Unity app and receive response (4 byte unsigned integer per command)
    client.push(buffer)
    # response = client.pull(buffer)

    # Evaluate response
    # print("Received response:")
    # print(response[0])
```

anjelomerte, 2024-04-30:
It simply was a problem of my network. For some reason it allowed messages from client to server but not the other way around. On another network it worked. Thanks a lot for your input!

anjelomerte, 2024-06-14:
Hey jdibenes!
In a more longterm perspective, communication from server to client will be pretty much necessary. Could you elaborate on how one could implement that? Thank you!

closed 2024-09-18 by anjelomerte

Zulex, 2024-10-01:
@anjelomerte 

Could you detail a bit more on how you achieved this?
I would like to send one of my Unity object coordinates together with the image frames (so they are synced in time), and I am deciding on what the best method would be.

anjelomerte, 2024-10-02:
As recommended by @jdibenes, I implemented polling from the client side. This means every x ms the client sends a message to the server and the server responds with a 4byte message (as with any other message). I only distinguished between two states though (0 and non-zero). However, I guess you could send a header with the number of bytes first and then evaluate the correspoding number of messages after that. Maybe there is a better way though. I also didn't explore @jdibenes 2nd suggestion:

> You can try using the HL2->PC 4-byte message queue by itself