jdibenes/hl2ss #102: Getting NullReferenceException when trying to build and run the Unity sample project

fmatulic, 2024-04-23:
I'm getting a NullReferenceException error in your BuildPostProcessor script when I try to build and run your sample Unity project. Here's the full trace:

```
NullReferenceException: Object reference not set to an instance of an object
BuildPostProcessor.AddCapability (System.Xml.XmlDocument xml, System.String name, System.String capability, System.String namespaceURI, System.Boolean append) (at Assets/Editor/BuildPostProcessor.cs:12)
BuildPostProcessor.OnPostprocessBuild (UnityEditor.BuildTarget target, System.String pathToBuiltProject) (at Assets/Editor/BuildPostProcessor.cs:33)
System.Reflection.RuntimeMethodInfo.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) (at <b11ba2a8fbf24f219f7cc98532a11304>:0)
```

I should specify that I'm using Unity 2022. Hopefully that's not an issue.

jdibenes, 2024-04-25:
Hello,
What Unity version are you using?

fmatulic, 2024-04-25:
Sorry for forgetting to mention that piece of info. I'm on Unity 2022, which I know isn't the version you used for the project.

jdibenes, 2024-04-29:
I tried building the Unity sample on Unity 2022.
Enabling Sync Capabilities in Project Settings > Player > Publishing Settings fixes the NullReferenceException error and the app seems to run correctly on the HoloLens.