jdibenes/hl2ss #105: Building the hl2ss app directly from VS sln failed

Walidkilas, 2024-04-30:
I 'am trying to build the server app and deploy to hololens 2  but i get this error 
1>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VisualStudio\v17.0\AppxPackage\Microsoft.AppXPackage.Targets(3069,5): error APPX0702: Payload file 'C:\Users\walid\Desktop\hl2ss\hl2ss\ARM64\Release\hl2ss\hl2ss.winmd' does not exist.
1>Done building project "hl2ss.vcxproj" -- FAILED.

jdibenes, 2024-04-30:
copy hl2ss.winmd from the etc folder to the ARM64\Release\hl2ss folder then try building again

Walidkilas, 2024-05-01:
Works fine thank you ! I have a new goal in mind and I'm wondering if you could assist me. I'm looking to develop a UWP application using Unity, which would run on my computer utilizing holographic remote. I'm considering integrating the player application https://github.com/microsoft/MixedReality-HolographicRemoting-Samples/tree/main/player/sample with your C++ solution to create a single application that can run on HoloLens 2, acting as both a server and holographic remote player simultaneously. What do you think about this idea? Do you think it's achievable, and if so, how could we make it work effectively?
The idea is to be able to preform a marker tracking with application that runs on my computer

jdibenes, 2024-05-01:
I'm not familiar with holographic remoting, but it may be possible to use the hl2ss DLL (unity plugin) and just call InitializeStreamsOnUI in the player application.

Walidkilas, 2024-05-01:
Amazing Thank you i will try it !

Walidkilas, 2024-05-01:
So , i tried to add the header hl2ss.h in the player script and called the InitializeStreamsOnUIt(2U) function to stream  the pv camera, however since i am not really familiar with c++ dev in VS i am facing a problem after the build the application cannot find the dll , i made some research and i found a solution on how to force VS to copy the dll to the ouput direcory using post build event but it still give me this error 
The program '[5452] SamplePlayer.exe' has exited with code 3221225781 (0xc0000135) 'A dependent dll was not found'.
Is there a way to copy manually the dll to HoloLens or do you have an idea on how can force it to include the dll ? thank you i apricate your help !

jdibenes, 2024-05-01:
are you also including the eye tracking and scene understanding dlls?
the hl2ss dll depends on them

Walidkilas, 2024-05-02:
yes i integrated all dll like this 
[image]
[image]

jdibenes, 2024-05-02:
I gave it a try. You can just add the dlls to the VS project (right click the project > Add > Existing Item) and then right click each of the three dlls > Properties > Set content to Yes.
But the IntializeStreamsOnUI won't work, you have to use InitializeStreams but that function is not exported. Just add UNITY_EXPORT above it in plugin.cpp and then add it to hl2ss.h (using UNITY_IMPORT) and rebuild the plugin. Then call InitializeStreams at the end of the SetWindow method in SamplePlayerMain.cpp. Also add perceptionSensorsExperimental and backgroundSpatialPerception to the appxmanifest if you want to stream IMU data.

Walidkilas, 2024-05-02:
Works fine the streaming is smooth and the application runs perfectly ! Amazing work i appreciate your effort thank you very much for your help

closed 2024-05-02 by Walidkilas

BrightLabFirenze, 2024-10-24:
Sorry I'm trying to install your fork  but the VSStudio solution doesn't work.
Can you share an application to install directly, as appxbundle

Please

jdibenes, 2024-10-29:
@BrightLabFirenze https://github.com/jdibenes/MixedReality-HolographicRemoting-Samples-hl2ss/releases/tag/v1.0.33.0