jdibenes/hl2ss #106: Transforming Aruco marker pose estimation to unity world coordinate

Metacadx, 2024-05-02:
Hello , i used client stream pv to perform an  Aruco code detection and pose estimation and  i managed to send the data to  unity however i can't find a way to superpose a 3D model onto the Aruco marker , do you have any suggestion ? Thank you for your help !

jdibenes, 2024-05-02:
Hello, both the pv pose and aruco pose are expressed in right handed coordinates while unity uses left handed coordinates.
Are you performing right hand to left conversion before applying the final transform to the 3D model?

Metacadx, 2024-05-02:
yes of course but i 'am not sure if i am doing thinks correctly when applying the transformation, I ve been looking for a solution for a while but seems like there is a lot of matrix transformation , so i am looking for a simpler and less expensive solution for the hololens 2 and the way i am doing , i feel like i am reinventing the wheel , is there any build in method or a simple equation to get the absolute position of aruco code using the transform of the camera ?

jdibenes, 2024-05-03:
The equation is `aruco_point_world_h = aruco_point_h @ aruco_pose^T @ pv_extrinsics^-1 @ pv_pose` where `_h` is for homogeneous 3D coordinates, `aruco_pose` is from `solvePnP` converted to a homogeneous 4x4 matrix, `pv_extrinsics` is a 4x4 constant, and `pv_pose` is the 4x4 transform of the PV frame.
We have included a [demo](https://github.com/jdibenes/hl2ss/blob/bf7f844798c3ade88787aebf9f1884bf92f19e5a/viewer/sample_aruco_pv.py#L100) that demonstrates this conversion and the right to left hand conversion required for Unity (lines 100-119).

Metacadx, 2024-05-04:
Thank you for you answer , there is something strange happening i get this error message ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host , the IP adress is correct and i tried  do you know where this error comes from ?

Metacadx, 2024-05-04:
Also there is a weird windows appearing and disappearing real quick  in the HoloLens application i can't tell what is it and this was not happening in the older build

jdibenes, 2024-05-04:
For Unity 2022 enable Sync Capabilities in Project Settings > Player > Publishing Settings.
The library and binaries were updated recently, check that you have the latest version of the files (hl2ss.py, hl2ss.dll in the unity folder, etc.). Since the dll was updated, the unity project has to be rebuilt from scratch (delete the old build folder and create a new one).

jdibenes, 2024-05-04:
My bad, I was using unity 2022.3.24f1.
I see now there is no sync capabilities in 2022.3.20f1, so the capabilities are set correctly already.

Metacadx, 2024-05-04:
Okay i will try to upgrade the unity version ! Do you have any plan on creating unity sample client ?

jdibenes, 2024-05-04:
do you mean a prebuilt version like the main app?
If so, I have added it to the [latest release](https://github.com/jdibenes/hl2ss/releases/tag/v1.0.28.0).

Metacadx, 2024-05-05:
No i mean an actual unity scene that runs as a c# client just like the python client ( not in c++  )

jdibenes, 2024-05-05:
No.
There are no plans for a unity c# client.