jdibenes/hl2ss #114: Questions about the imu data

JadeGreened, 2024-06-17:
I finally get the imu data from my hololens! Thanks a lot~. But I notice that the imu data is a 4x4 matrix like this   

[[ 0.05868477 -0.9801954   0.18913764  0.        ]
 [-0.96467817 -0.00694191  0.26333955  0.        ]
 [-0.25681123 -0.19791096 -0.94598055  0.        ]
 [-0.12868184  0.15060404 -0.22494759  1.        ]]

Since I am new to imu-activity-recognition, I have no idea of what the rows and the columns mean. And the microsoft documentation did not explain the shape of the matrix. Could you please provide me with more detailed explanation or show me the relavant documentation, plz~. 
if the data is present like this:  
 Research Mode IMU
Accelerometer (m/s^2)
Gyroscope (deg/s)
Magnetometer  
Then why the matrix is four-dimensional？

anjelomerte, 2024-06-17:
How exactly are you getting this matrix, could you share your implementation? Looking at [https://github.com/jdibenes/hl2ss/blob/main/viewer/client_stream_rm_imu.py](url), 6 values (sensor_ticks, soc_ticks, x, y, z and temperature) are returned.

JadeGreened, 2024-06-17:
sry~ I forget the comment on the top of the script

JadeGreened, 2024-06-17:
problem solve :)

JadeGreened, 2024-06-17:
But if I want to have three types of data together  
# Port
# Options:
# hl2ss.StreamPort.RM_IMU_ACCELEROMETER
# hl2ss.StreamPort.RM_IMU_GYROSCOPE
# hl2ss.StreamPort.RM_IMU_MAGNETOMETER  
Should I open 3 client for each ports?

jdibenes, 2024-06-21:
Hello,
Yes. The data volume for these channels is low so multithreading should work or you can use the multiprocessing utilities like in the samples.