jdibenes/hl2ss #116: co-ordinate system and timestamp for Extended Eye Tracking

aipsita, 2024-07-01:
Hello. I am new to eye tracking technology. I want to analyze eye tracking data captured from the Hololens 2 using client_stream_eet.py. The data I record is saved as shown below.
[image]

I am trying to convert the eye gaze co-ordinates (origin, direction) from device coordinate to world coordinate system. Is this the correct approach?
[image]

Also I wanted to confirm if the timestamp resolution is 100 ns.

Thank you so much.

jdibenes, 2024-07-02:
Hello,
device to world conversions:
origin: `[x,y,z,1]@pose`
direction: `[x,y,z]@pose[:3,:3]`
yes, timestamp resolution is 100 ns

aipsita, 2024-07-02:
Thank you so much.

closed 2024-07-02 by aipsita