jdibenes/hl2ss #117: Question about the IMU calibration matrices

zijianh4, 2024-07-04:
Hi,
Thanks for the awesome work! I am currently trying to calibrate the IMU sensors with cameras. However, I find that the extrinsic parameters of the accelerator is 
```
[[-0.9814974  -0.15415965  0.1135679   0.        ]
 [ 0.07882602  0.21521354  0.9733805   0.        ]
 [-0.17449735  0.9643225  -0.19907974  0.        ]
 [ 0.01642743  0.04962904  0.11200297  1.        ]]
```
and the extrinsic parameters of the gyroscope is 
```
[[ 0.9795797   0.16419245  0.11603672  0.        ]
 [-0.07659374 -0.22886042  0.9704413   0.        ]
 [ 0.18589534 -0.95951223 -0.2116109   0.        ]
 [ 0.          0.          0.          1.        ]]
```
and thus I have two questions about this
1) why are these two sensors have different extrinsic matrices?
2) why there is no data in the translation part of the gyroscope's extrinsic matrix?
Thanks!

jdibenes, 2024-07-15:
Hello,
I don't know, sorry. I tried looking in the HoloLens2ForCV repo but was unable to find any relevant information.
I assume 1) is due to them being two separate physical devices and 2) since it is used to measure orientation changes, translation is not used to transform rotation matrices, but this is just speculation.

zijianh4, 2024-08-13:
> Hello, I don't know, sorry. I tried looking in the HoloLens2ForCV repo but was unable to find any relevant information. I assume 1) is due to them being two separate physical devices and 2) since it [...]

Thanks for your reply. I personally also hold this idea. However, if this is the case, then it seems that we can not use this IMU data with cameras especially the gyroscope part. This is because the translation part is important when we use the gyroscope readings to do pose estimation such as SLAM algorithm.