jdibenes/hl2ss #120: VLC stereo depth estimation

Zulex, 2024-07-17:
Hi, first of all thanks for this great repo, it is super helpful! 

I am trying to do depth estimation of a single keypoint using both front VLC camera's. I am following the sample (sample_stereo_rectify.py), to undistort and rectify both images. Afterwards for testing purposes I manually pick a keypoint in both images (Left being 1 and right being 2 if I understand correctly) and use cv2.triangulatepoints. For P1 and P2 I use the saved stereo_rectification file created by the same script. 

So I follow this procedure:

1. Load undistorted and rectified images
2. Load calibration and rectification files
3. Create variables containing the keypoints : 
pts1 = [[x_l , y_l]] & pts2 = [[x_r , y_r]]
4. Triangulate using 
cv2.triangulatePoints(stereo_rectification.P1, stereo_rectification.P2, pts_1, pts_2)
5. Converting 4D matric to 3D points: 
points_3d = points_4d_homogeneous[:3] / points_4d_homogeneous[3]
6. Get T1 from the left camera to use as reference point: 
T = stereo_rectification.P1[:,3]
7. Calculate distance from reference point to keypoint: 
distance = np.sqrt(np.sum((points_3d - T)**2))

However I get a value of  ~3 (I assume in mm), while I was expecting around 20cm. What should I being doing differently? 
Should I be creating image_to_world matrixes to use as P1 and P2 and then use (0,0,0) as reference point?

jdibenes, 2024-07-19:
Hello,
For triangulation, you have to use unrectified coordinates. If you want to use rectified coordinates you have to undo the rectification using the Q matrix like in OpenCV. See stereoRectify() in https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6

Zulex, 2024-07-22:
Thanks for the clarification @jdibenes, I think I had misinterpreted the P1 output, as it reads that it is the projection matrix in the new rectified frames. I couldn't get it too work with unrectification and decided too directly try to triangulate before rectifying, as I can find my keypoints without the need of rectification.

However, this also doesn't seem to work. I think I am misunderstanding the intrinsics/extrinsic values and mixing up row and column major operations. I thought this process would've been fairly, and tried many sanity checks and flipped everything around; but all the numbers still don't make sense. 

I think in theory I should be getting 'correct' position by doing this: 

```python
# Image to rignode matrix
Pl = hl2ss_3dcv.camera_to_rignode(calibration_l.extrinsics) @ hl2ss_3dcv.image_to_camera(calibration_l.intrinsics)
Pr = hl2ss_3dcv.camera_to_rignode(calibration_r.extrinsics) @ hl2ss_3dcv.image_to_camera(calibration_r.intrinsics)

# Transpose to get the translation in the last column
Pl = Pl.T
Pr = Pr.T

#make it a 3x4 matrix that can be used in triangulatePoints
Pl= Pl[:3,:4]
Pr= Pr[:3,:4]
```

After which I perform triangulate points on the undistorted images (of which I kept the original rotation): 

```python
points_4d_homogeneous  = cv2.triangulatePoints(Pl, Pr, pts_left_T, pts_right_T)
points_3d = cv2.convertPointsFromHomogeneous(points_4d_homogeneous.T)
T = (0,0,0)
distance = np.sqrt(np.sum((points_3d - T)**2))
```

My keypoints are in cv2 format, in which (0,0) is the top left, and they are not normalized (so every pixel is an integer). The distorted images have been tried too. 

Any pointers in the right direction would be hugely appreciated!

jdibenes, 2024-07-23:
Here is an example for triangulation on unrectified images:
```
    # Get camera calibrations -------------------------------------------------
    calibration_lf = hl2ss_3dcv.get_calibration_rm(host, port_left,  calibration_path)
    calibration_rf = hl2ss_3dcv.get_calibration_rm(host, port_right, calibration_path)

    rotation_lf = hl2ss_3dcv.rm_vlc_get_rotation(port_left)
    rotation_rf = hl2ss_3dcv.rm_vlc_get_rotation(port_right)

    K1, Rt1 = hl2ss_3dcv.rm_vlc_rotate_calibration(calibration_lf.intrinsics, calibration_lf.extrinsics, rotation_lf)
    K2, Rt2 = hl2ss_3dcv.rm_vlc_rotate_calibration(calibration_rf.intrinsics, calibration_rf.extrinsics, rotation_rf)

    # Get projection matrices (X_3DH.T @ Rt1 @ K1) -----------------------------
    P1 = hl2ss_3dcv.rignode_to_camera(Rt1) @ hl2ss_3dcv.camera_to_image(K1)
    P2 = hl2ss_3dcv.rignode_to_camera(Rt2) @ hl2ss_3dcv.camera_to_image(K2)

    # Using undistorted, rotated, unrectified images --------------------------
    # lf_u = cv2.remap(data_left.payload,  calibration_lf.undistort_map[:, :, 0], calibration_lf.undistort_map[:, :, 1], cv2.INTER_LINEAR)
    # rf_u = cv2.remap(data_right.payload, calibration_rf.undistort_map[:, :, 0], calibration_rf.undistort_map[:, :, 1], cv2.INTER_LINEAR)
    # lf_ru = hl2ss_3dcv.rm_vlc_rotate_image(lf_u, rotation_lf)
    # rf_ru = hl2ss_3dcv.rm_vlc_rotate_image(rf_u, rotation_rf)
    # image_l = hl2ss_3dcv.rm_vlc_to_rgb(lf_ru)
    # image_r = hl2ss_3dcv.rm_vlc_to_rgb(rf_ru)
    # image = np.hstack((image_l, image_r))

    # Get two correspondences from checkerboard pattern -----------------------
    p1 = np.array([[523/2,        575/2       ], [249/2, 249/2]]) # Image L points [[x1, x2],[y1, y2]]
    p2 = np.array([[1321/2 - 480, 1373/2 - 480], [245/2, 243/2]]) # Image R points [[x1, x2],[y1, y2]]

    # Triangulate -------------------------------------------------------------
    points = cv2.triangulatePoints(P1[:,:3].T, P2[:,:3].T, p1, p2)

    x1 = points[:3, 0] / points[3, 0]
    x2 = points[:3, 1] / points[3, 1]
    
    print(np.linalg.norm(x1 - x2)) # Measure distance between two 3D points in checkerboard pattern (known distance)
```

Zulex, 2024-07-23:
Thanks, that did the trick!