jdibenes/hl2ss #129: Disable Auto Focus

nirajkark07, 2024-08-11:
Hello,

Is there a way to disable the autofocus in the client_stream_pv.py demo? I am looking to calibrate the HL2 camera with an external camera. Therefore, I require constant intrinsic parameters.

Thank you!

jdibenes, 2024-08-12:
Hello,
After start_subsystem_pv, add
```
client_rc = hl2ss_lnm.ipc_rc(host, hl2ss.IPCPort.REMOTE_CONFIGURATION)
client_rc.open()
client_rc.wait_for_pv_subsystem(True)
client_rc.set_pv_focus(hl2ss.PV_FocusMode.Manual, hl2ss.PV_AutoFocusRange.Normal, hl2ss.PV_ManualFocusDistance.Infinity, focus_value, hl2ss.PV_DriverFallback.Disable)
client_rc.close()
```
`focus_value` must be in the range [170,10000], 2000 is usually good.

nirajkark07, 2024-08-17:
Awesome! Closing this thread now. Thanks. :)

closed 2024-08-17 by nirajkark07