jdibenes/hl2ss #138: Compatibility Issue with Holographic Remoting and Camera Access in Unity Scene

vincenzocivale, 2024-09-06:
Hello,

I am trying to use the hl2ss plugin within my Unity project via Holographic Remoting. My goal is to access the HoloLens cameras in real-time, as the latest version of Holographic Remoting should, in theory, allow for camera access on the device. However, I am encountering an issue where the Python script is unable to detect the device when remoting is active.

Despite successfully connecting via Holographic Remoting and rendering the scene on my HoloLens, the Python client fails to communicate with the device, specifically when trying to access the front RGB or depth cameras.

Any guidance or support would be greatly appreciated. Thank you for your work on this project!

jdibenes, 2024-09-07:
Hello,
Try changing line 8 of the hl2ss.cs script to `#if true`.
If that does not work, undo that change and you could try building the Holographic Player from https://github.com/microsoft/MixedReality-HolographicRemoting-Samples with hl2ss support as described in https://github.com/jdibenes/hl2ss/issues/105 and https://github.com/jdibenes/hl2ss/issues/109.

vincenzocivale, 2024-09-09:
Thank you very much for the suggestion! After applying the modification, I encountered the error DllNotFoundException: hl2ss assembly:<unknown assembly> type:<unknown type> member:(null). I believe I need to recompile the plugin to work on non-ARM64 systems, but I don't have much experience with this, so it might take some time.

I will update the issue once I've tested it.

P.S. If anyone already has a version compatible with Windows x86 systems or has suggestions on how to proceed, I would greatly appreciate it.

vincenzocivale, 2024-09-11:
Sorry, I've followed the instructions of #109, but I have this error that I don't know how to fix:

Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	LNK1104	cannot open file 'hl2ss.lib'	SamplePlayer	C:\Users\cical\Documents\GitHub\Repositories\RemoteHoloSenseLink\MixedReality-HolographicRemoting-Samples\player\sample\LINK	1		

I've added the DLL from plugin_unity_1.0.33.0 and the lib file from plugin_uwp_1.0.33.0.

[image]

jdibenes, 2024-09-11:
did you set `Set content = Yes` for all three dlls?

vincenzocivale, 2024-09-11:
Yes I set Content ti "Yes" for hl2ss.dll and for Microsoft dll too from Visual Studio 2022

jdibenes, 2024-09-11:
hard to tell what's wrong
try this fork https://github.com/jdibenes/MixedReality-HolographicRemoting-Samples-hl2ss
it's already configured and runs
just build as Release ARM64

vincenzocivale, 2024-09-13:
Apologies for the delayed response; I was trying to resolve the issue on my own, to not disturb again you. I tried using the fork you suggested, but unfortunately, I'm still encountering the same error:

`Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	LNK1112	module machine type 'x64' conflicts with target machine type 'ARM64'	SamplePlayer	C:\Users\cical\Documents\GitHub\Repositories\RemoteHoloSenseLink\MixedReality-HolographicRemoting-Samples\player\sample\SamplePlayer.dir\Release\pch.obj	1`
	
From what I undestand, the hl2ss libraries should be compiled exclusively for ARM64, so I'm not sure why this error is occurring. Any suggestions on what might be causing this?

vincenzocivale, 2024-09-13:
I've resolved the issue by building the project on a different PC. I suspect that the error might have been related to the version of Visual Studio on my original machine. Thank you once again for the support, and I want to commend you on the excellent work on this project!

closed 2024-09-13 by vincenzocivale