jdibenes/hl2ss #142: Is it possible to build the server app from scratch and intergrate it to some other Unity App?

zijianh4, 2024-09-26:
Hi, 
I am currently trying to run the hl2ss app in 2d version with another 3D app which uses the Anchor service of Microsoft. However, it seems that one of them always quit because of the HoloLens 2 resource limitation. Therefore, I wonder if there is a way to intergrate the hl2ss server app to another Unity project. Thanks!

jdibenes, 2024-09-26:
Hi,
If you have access to the Unity project you can try adding the hl2ss plugin to it. See hl2ss_unity for an example.

zijianh4, 2024-09-30:
> Hi, If you have access to the Unity project you can try adding the hl2ss plugin to it. See hl2ss_unity for an example.

Thanks. Sorry for another question. If I want to build the plugin directly from source in another project. Do you know how to do it. I notice that there is one guide for building the whole app directly from source and mention the plugin but I am not sure if I need to follow the same steps. Thanks!

jdibenes, 2024-10-01:
The plugin depends on winrt code generated when building the app so it may be complicated (the VS solution referenced by the guide builds the app and then the plugin).
What's your setup?
Are you trying to add the C++ files to Unity and use IL2CPP?

zijianh4, 2024-10-02:
> The plugin depends on winrt code generated when building the app so it may be complicated (the VS solution referenced by the guide builds the app and then the plugin). What's your setup? Are you trying to add the C++ files to Unity and use IL2CPP?

Hi, basically I want to capture images and IMU data by adding this plugin into the Microsoft Anchor App https://github.com/Azure/azure-spatial-anchors-samples, I tried to extract the folders and files into the Assets folder of the Anchor App but I receive the compilation error which says "Assets\Scripts\RemoteUnityScene.cs(7,30): error CS0234: The type or namespace name 'Toolkit' does not exist in the namespace 'Microsoft.MixedReality' (are you missing an assembly reference?)". Do you know how I can resolve this error? Thanks!
[image]

zijianh4, 2024-10-02:
> The plugin depends on winrt code generated when building the app so it may be complicated (the VS solution referenced by the guide builds the app and then the plugin). What's your setup? Are you trying to add the C++ files to Unity and use IL2CPP?

Hi, I extracted the plugin files into the Assets folder as required and the Anchor App https://github.com/Azure/azure-spatial-anchors-samples can run with the Plugin on HoloLens 2. However, I keep getting this error which seems saying that the portal cannot transfer the data. Could you please help me to resolve this issue? Thanks! As for the context, I run the `./viewer/simple_recorder.py` to receive the leftfront and right front cameras. 
[image]

jdibenes, 2024-10-02:
Did you add the Hololens2SensorStreaming.cs script to the Main Camera?
The plugin requires some setup when added to a Unity project.
See "Adding the plugin to an existing project" under "Unity plugin" in the readme for details.

zijianh4, 2024-10-02:
> Did you add the Hololens2SensorStreaming.cs script to the Main Camera? The plugin requires some setup when added to a Unity project. See "Adding the plugin to an existing project" under "Unity plugin" in the readme for details.

Hi,
Yes, I follow the set up under "Unity plugin" and I think it is a little bit strange that although it says recording start later it will report the timeout issue. Previously when I try collecting the data directly with the hl2ss app, if it does not connect successfully the recording starts line will not show. Here is also a screenshot of the Package.appxmanifest after I build the App.
[image]

jdibenes, 2024-10-03:
>I think it is a little bit strange that although it says recording start later it will report the timeout issue. Previously when I try collecting the data directly with the hl2ss app, if it does not connect successfully the recording starts line will not show. Here is also a codeblock of the Package.appxmanifest after I build the App.

Check that the Unity App has access to:
  - Camera (Settings -> Privacy -> Camera).
  - Eye tracker (Settings -> Privacy -> Eye tracker).
  - Microphone (Settings -> Privacy -> Microphone).
  - User movements (Settings -> Privacy -> User movements).

zijianh4, 2024-10-03:
> Check that the Unity App has access to:
>
> * Camera (Settings -> Privacy -> Camera).
> * Eye tracker (Settings -> Privacy -> Eye tracker).
> * Microphone (Settings -> Privacy -> Microphone).
> * User [...]

Hi,
I am not sure if I understand correctly, but I can only find the Project settings under Edit 
[image]
and I follow the setup here to open the access
[image]
Is this correct? Thanks!

jdibenes, 2024-10-03:
Settings from the Start menu on the HoloLens, which opens when you tap your wrist.

zijianh4, 2024-10-03:
> Settings from the Start menu on the HoloLens, which opens when you tap your wrist.

Hi,
Yes. I turn on all the permissions and the error still exists.