jdibenes/hl2ss #147: Video Stream Latency Issue

TheGoncaloSilva, 2024-10-16:
First of all, thank you for the work you did in this library.
I'm using the Hololens 2 for a real-time scenario where i need to have processing of the video camera done by another device. Using the  compiled appxbundle, i managed to get a very good latency of 180ms on 1504x846@60, however, once i do the same test in Unity, i get closer to 600ms. I've tested in my Unity project and in the provided sample, with equal results. Also, the visual profiler clearly displays that the device has plenty of RAM and no stutters. Did you encounter something similar?

jdibenes, 2024-10-19:
Hi,
Yes, currently there is no solution other than reducing resolution and framerate.

TheGoncaloSilva, 2024-10-21:
Thank you for your response. That's what i figured, i'll try to find a way to sideload the Unity project and hl2ss, to see if i can reduce the latency

closed 2024-10-21 by TheGoncaloSilva