jdibenes/hl2ss #149: hl2ss project appxbundle fails to run after HoloLens 2 reset and OS update in October

vincenzocivale, 2024-10-27:
**Problem Description**:  
After performing a factory reset on my HoloLens 2 in October and updating it to the latest OS version, the `.appxbundle` for the **hl2ss** project no longer functions correctly. Before the reset, the application was running without issues.

**Steps to Reproduce**:
1. Perform a factory reset on HoloLens 2 and install the latest OS update (as of October 2023).
2. Install the application using the `.appxbundle` for the **hl2ss** project.
3. The application installs correctly, but upon launch:
   - A window appears with the text "Update".
   - After this screen, the application window remains open but does not proceed.

**Expected Behavior**:  
The application should launch and function as it did before the reset, without freezing on the update screen.

**Additional Observations**:
- From the HoloLens 2 WebView, it appears that the **hl2ss** application process is not running.
- There are no specific error messages; the app just seems to not start correctly.

**Technical Details**:
- **Device**: HoloLens 2
- **OS Version**: Latest version available in October 2023
- **hl2ss Version**: latest
- **Visual Studio Version and Build Configurations**: [specify if possible, e.g., ARM64, Release]

**Troubleshooting Attempts**:
- I’ve tried reinstalling the app and restarting the device without success.

**Request**:  
Has anyone experienced similar issues or knows of a solution? Any advice would be greatly appreciated.

jdibenes, 2024-10-27:
did you re-enable research mode after the reset?

vincenzocivale, 2024-10-27:
Yes, I've re-enabled research mode.

vincenzocivale, 2024-10-29:
I'm trying to build your fork of MixedReality HolographicRemoting, But appear the error hl2ss.pdb not found. Do you knwo how to resolve the issue?

jdibenes, 2024-10-29:
The pdb file is in the plugin_uwp zip folder, under releases. But, if VS is asking for it then the application is crashing. Also, the remote player must be built as Release ARM64.
I was able to run the hl2ss app successfully on the latest OS build (22621.1402 October 2024).
Can you verify your OS build number and that Developer Mode/Device Portal are enabled and that the hl2ss app permissions are set correctly?