jdibenes/hl2ss #154: Aligning SI gaze and PV stream - sample_si_pv.py example script not working

bluejaelly, 2025-01-04:
Hello! Thanks for this amazing package.

I am trying to project gaze coordinate from SI stream onto the camera image frames from PV stream. I believe the sample_si_pv.py script is an example of what I'm trying to accomplish.

However, when I run the sample_si_pv.py, the run ends without any output (i.e., no error messages, no logs, no cv2.imshow windows). I was wondering if this script is still working and whether it is accomplishing what I intend to do.

Any help would be awesome, thank you!

jdibenes, 2025-01-04:
Hi,
The sample_si_pv script also uses spatial mapping to determine depth.
Can you check if client_ipc_sm.py runs correctly on your machine?

bluejaelly, 2025-01-04:
Hi,

Thanks so much for getting back to me!

I tried running client_ipc_sm.py, and it seems to print the following line:
print(f'Task {index}: surface id {id_hex} @ {timestamp} has {mesh.vertex_positions.shape[0]} vertices {mesh.triangle_indices.shape[0]} triangles {mesh.vertex_normals.shape[0]} normals')

However, nothing seems to be visualized.

On visual studio, I see "SM: Client connected".

However, my guess is, there is still an issue with how I am perhaps setting up spatial mapping for hl2ss...? Are there specific hl2ss-related settings I should look into?

Thanks again!

jdibenes, 2025-01-04:
No, there are no specific settings. It should work as is.
If the script is printing that line then spatial mapping download should be working.
If you comment out lines 72 to 79 the script should print:
`Observed N surfaces`
and then print `Task 0: ...` to `Task N-1: ...` with non-zero vertices, triangles, and normals.
Also, what Open3D version are you using?
You can use `pip show open3d`.

bluejaelly, 2025-01-05:
Thanks!

I did comment out lines 72 to 79 from `client_ipc_sm.py`. I get `Observed 8 surfaces`. Then something like `Task 0: surface id 9484e212903ed946905a320310bc0391 @ 133805416996670260 has 18 vertices 16 triangles 18 normals`.

Additionally, according to `pip show open3d`, I am using version 0.18.0. I believe this is the latest version, since I just ran `pip install open3d`.

Finally, I tried running `sample_si_pv.py` again. I left in the debug logs:

77 print(inspect.getframeinfo(inspect.currentframe()).lineno)
78 sm_manager = hl2ss_sa.sm_manager(host, triangles_per_cubic_meter, mesh_threads)
79 print(inspect.getframeinfo(inspect.currentframe()).lineno)
80 sm_manager.open()
81 print(inspect.getframeinfo(inspect.currentframe()).lineno)
82 sm_manager.set_volumes(volumes)
83 print(inspect.getframeinfo(inspect.currentframe()).lineno)
84 sm_manager.get_observed_surfaces()
85 print(inspect.getframeinfo(inspect.currentframe()).lineno)

This printed up to 83, but did not print 85. Hence, my guess is, `sm_manager.get_observed_surfaces()` is the line that is causing the program to stop running. I do not see any error messages, but the code stops running.

Please let me know if there's anything else I should try!

jdibenes, 2025-01-05:
Try resetting the map, from the device portal > system > map manager.

bluejaelly, 2025-01-05:
Thanks again! I tried resetting the map, but the code still doesn't seems to reach `sm_manager.get_observed_surfaces()`...

Any other suggestions? :))

jdibenes, 2025-01-05:
What Python version are you using?
I've only tested the scripts on 3.9.12.

bluejaelly, 2025-01-05:
Looks like I'm using Python 3.9.21.

What I can try is to (1) reset the HoloLens; (2) use Python 3.9.12; and (3) do git clone to use your project exactly as it is right now (I made some edits, though I don't think it's significant enough to impact performance).

Afterwards, if there are still issues, I'll circle back. Thanks!

bluejaelly, 2025-01-06:
Hi,

I tried (1) resetting the HoloLens completely; (2) using Python 3.9.12; and (3) git cloning and using the entire hl2ss package AS IS. However, I am still seeing the same exact behavior / error I've written in my previous comments...

To recap:
1. `client_ipc_sm.py` runs fine:
Observed 14 surfaces
Task 0: surface id 932e86e8f25f5e40aca5675bd9d74b79 @ 133806268188707433 has 62 vertices 81 triangles 62 normals
...
Though when I uncomment lines 72 - 79, it says observed 1 surface, does not visualize anything, and the code just exits... 

2. `client_stream_pv.py` runs fine. Gives the proper video frame + pose data when setting mode to `mode = hl2ss.StreamMode.MODE_1`

3. `client_stream_si.py` runs fine. All the logs look correct and it can correct tell if my left or right hand appears/disappears.

4. Then I tried to run `sample_si_pv.py`. Again, the code runs until right before the line `sm_manager.get_observed_surfaces()`, but then exits without any error messages as soon as it reaches this exact line...

As a result, I still think there are issues with either the `hl2ss_sa.py` file or the `open3d` package.

Is there anything I'm missing that I can try? Thanks again!

jdibenes, 2025-01-07:
I think I have managed to replicate your issue.
I upgraded numpy to version 2.0.2 and the `client_ipc_sm.py` and `sample_si_pv.py` scripts now fail exactly as you describe.
The following configuration should work, at least on Windows 10 and Python 3.9.12:
- numpy 1.26.4
- opencv-python 4.10.0.84
- open3d 0.18.0

bluejaelly, 2025-01-07:
Aha! I tried **downgrading numpy from 2.0.2 to 1.26.4**. Now both `client_ipc_sm.py` and `sample_si_pv.py` executes properly! 
Thanks so much for looking into this! 

Hope anyone having the same issue can also refer to this :))

closed 2025-01-07 by bluejaelly