jdibenes/hl2ss #155: [WinError 10054] an existing connection was forcibly closed by the remote host

Secret-Lin, 2025-01-16:
@jdibenes , thanks for your great work.
Recently, I have encountered a strange issue. On one Windows 11 computer, I can run all the code, but when I use another Windows 11 computer, I am unable to run code that calls pv. However, I can run other scripts, such as client_stream_rm_depth_ahat.py. The virtual environments on both Windows 11 machines were copied using Conda. The error infomation is shown below:
```
Traceback (most recent call last):
  File "E:\UnityProjects\hl2ss\viewer\client_stream_pv.py", line 91, in <module>
    client.open()
  File "E:\UnityProjects\hl2ss\viewer\hl2ss.py", line 1568, in open
    self.get_next_packet()
  File "E:\UnityProjects\hl2ss\viewer\hl2ss.py", line 1571, in get_next_packet
    data = super().get_next_packet()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\UnityProjects\hl2ss\viewer\hl2ss.py", line 781, in get_next_packet
    return self._client.get_next_packet()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\UnityProjects\hl2ss\viewer\hl2ss.py", line 403, in get_next_packet
    self._unpacker.extend(self._client.recv(self._chunk_size))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\UnityProjects\hl2ss\viewer\hl2ss.py", line 291, in recv
    chunk = self._socket.recv(chunk_size)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [WinError 10054] an existing connection was forcibly closed by the remote host.
```
By the way, using hl2ss_unity, it can successfully run samle_aruco_py.py, which also calls pv. That's strange.

jdibenes, 2025-01-17:
Yes, it is strange.
That error usually happens when the PV subsystem is off or the server received an invalid configuration.
Maybe there was some change to the client_stream_pv.py script?
does client_stream_pv.py work with hl2ss_unity?

Secret-Lin, 2025-01-23:
> Yes, it is strange. That error usually happens when the PV subsystem is off or the server received an invalid configuration. Maybe there was some change to the client_stream_pv.py script? does client_stream_pv.py work with hl2ss_unity?

Sorry for the late reply, I was on vacation and could not test.
After testing, client_stream_pv.py does not work. Actually, all py or cpp request for pv msg fail in one win11 computer (A), and work fine in another win11 computer (B).
After checking code,  virtual environments and firewall, these factors can be excluded.
The possible issue is the Windows 11 Network Adapter and its driver version. In computer A, it used Intel(R) Wi-Fi 6 AX201 and latest driver (2024/5/1). In computer B, it used Intel(R) Wi-Fi 6 AX211 and latest driver (2024/6/3). As computer A is belongs to lab and I can not change the driver version, I am no longer using it.
Thanks for your reply.

Secret-Lin, 2025-12-23:
For using VS2022 and the new version of hl2ss, the problem is solved.