jdibenes/hl2ss #156: sample_2d_to_3d_segmentation_pv.py -> no module named 'mmdet.core'

sreeharshaparuchur1, 2025-01-25:
Hey,

Thanks for this terrific work!

I face the following error when trying to run ```sample_2d_to_3d_segmentation_pv.py```

```
python sample_2d_to_3d_segmentation_pv.py
/home/kneepolean/hololens/hl2/lib/python3.10/site-packages/torch/cuda/__init__.py:82: UserWarning: CUDA initialization: CUDA unknown error - this may be due to an incorrectly set up environment, e.g. changing env variable CUDA_VISIBLE_DEVICES after program start. Setting the available devices to be zero. (Triggered internally at  ../c10/cuda/CUDAFunctions.cpp:112.)
  return torch._C._cuda_getDeviceCount() > 0
Traceback (most recent call last):
  File "/home/kneepolean/hololens/hl2ss/viewer/sample_2d_to_3d_segmentation_pv.py", line 22, in <module>
    from mmdet.core import INSTANCE_OFFSET
ModuleNotFoundError: No module named 'mmdet.core'
```
I have tried installing mmcv==2.1.0 using pip, using mim. I've even tried installing mmdet using pip and then building it from source. 

Please advise on what I am missing out on doing. 

Additionally, I look to use this method to segment a specified class (out of only 2 classes, a bone model and a robot manipulator) in that using the RGB input of the camera, I want to send that and depth data to offboard compute, perform image segmentation, reproject those masks into 3d and view it on the HL2. The client_stream_pv.py file seems to have a very high latency but is that an accurate representation of latency between the HL2 and a system connected over Wifi? Moreover, what possibly optimisations could you suggest for my task.

I'd sincerely appreciate any insight you could give!

Thanks in Advance

jdibenes, 2025-01-26:
Hi,
Try using mmdetection v2.25.1.
Are you using the Unity plugin? Currently, PV streaming has high latency for resolutions higher than 1280x720 when using the plugin in Unity apps.

sreeharshaparuchur1, 2025-02-13:
Hello,

Thank you for responding soon, I was trying out several things since then and taken care to use the following versions of libraries to run the segmentation code:

mmcv-full                 1.6.0                    pypi_0    pypi
mmdet                     2.25.1                   pypi_0    pypi
mmengine                  0.10.6                   pypi_0    pypi
numpy                     1.25.2                   pypi_0    pypi
open3d                    0.19.0                   pypi_0    pypi
opencv-python             4.11.0.86                pypi_0    pypi
torch                     1.11.0                   pypi_0    pypi
torchaudio                0.11.0                py310_cpu    pytorch
torchvision               0.12.0                py310_cpu    pytorch

However, it seems like mmdet 2.25.1 isn't compatible with the way that the code is currently written, for example: 
`from mmdet.registry import DATASETS` is not valid, I had to change it to `from mmdet.datasets import DATASETS`.

## The issue

Using [this config](https://github.com/open-mmlab/mmdetection/blob/master/configs/mask2former/mask2former_swin-t-p4-w7-224_lsj_8x2_50e_coco-panoptic.py) with the [SWIN T](https://github.com/open-mmlab/mmdetection/blob/main/configs/mask2former/README.md) model in the mmdetection zoo park.

I have made multiple changes to the files that lie in function calls to finally face an error due to the discrepancy in model state names generated by the config passed to 'init_detector' from mmdet.apis with that of the state dictionary keys in the pretrained model.

For example:

The keys desired by the model generated by the config:
`
panoptic_head.pixel_decoder.encoder.layers.0.attentions.0.value_proj.weight, panoptic_head.pixel_decoder.encoder.layers.0.attentions.0.value_proj.bias, panoptic_head.pixel_decoder.encoder.layers.0.attentions.0.output_proj.weight, panoptic_head.pixel_decoder.encoder.layers.0.attentions.0.output_proj.bias, panoptic_head.pixel_decoder.encoder.layers.0.ffns.0.layers.0.0.weight, panoptic_head.pixel_decoder.encoder.layers.0.ffns.0.layers.0.0.bias, panoptic_head.pixel_decoder.encoder.layers.0.ffns.0.layers.1.weight, panoptic_head.pixel_decoder.encoder.layers.0.ffns.0.layers.1.bias,
`

But these are the keys in the SWIN T model state dictionary:

`
panoptic_head.pixel_decoder.encoder.layers.0.self_attn.value_proj.weight, panoptic_head.pixel_decoder.encoder.layers.0.self_attn.value_proj.bias, panoptic_head.pixel_decoder.encoder.layers.0.self_attn.output_proj.weight, panoptic_head.pixel_decoder.encoder.layers.0.self_attn.output_proj.bias, panoptic_head.pixel_decoder.encoder.layers.0.ffn.layers.0.0.weight, panoptic_head.pixel_decoder.encoder.layers.0.ffn.layers.0.0.bias, panoptic_head.pixel_decoder.encoder.layers.0.ffn.layers.1.weight, 
panoptic_head.pixel_decoder.encoder.layers.0.ffn.layers.1.bias,
`

As we can see, they differ in the way they reference 'ffn' and 'self_attn'. There are several more mismatches like this.

Thus, it would be helpful if you could upload and share the config file and pretrained model weights that you use to run the above script. If you could verify that the versions of the libraries that I'm using also match up with you that would be beyond helpful.

Thanks in Advance!

jdibenes, 2025-02-13:
Here are the config and weights https://files.catbox.moe/f11t79.rar

Versions:
mmcv-full 1.6.1
mmdet 2.25.1
mmengine not installed
numpy 1.26.4
open3d 0.18.0
opencv-python 4.10.0.84
torch 1.12.1+cu116
torchaudio 0.12.1+cu116
torchvision 0.13.1+cu116
Python 3.9.12

sreeharshaparuchur1, 2025-02-24:
Hi,

Thank you so much for the prompt reply. With the checkpoint and config you provided, I was toying around with segmenting objects (such as bone models, a robot manipulator) in the HoloLens FoV and registering them to a local frame to build a pointcloud of the scene. 

However, even after checking version dependencies, I could only run this code on the "cpu" device, as when device is set to "cuda"/"cuda:0", I face runtime errors - below is a snippet of the bottom of the runtime error that I get... i'm not sure what's wrong as i've matched all versions with what you've outlined. If you could help me out, that would be great, if not, that's okay and i'll close the issue.

```
 template <int output_vec_size>
  C10_DEVICE Array<arg_t, output_vec_size> global_reduce(Array<arg_t, output_vec_size> value, Array<arg_t, output_vec_size> *acc, char* shared_memory) const {
    using arg_vec_t = Array<arg_t, output_vec_size>;
    using out_ptr_vec_t = Array<out_scalar_t*, output_vec_size>;
    using offset_vec_t = Array<uint32_t, output_vec_size>;

    arg_vec_t* reduce_buffer = (arg_vec_t*)cta_buf;
    uint32_t output_idx = config.output_idx<output_vec_size>();
    offset_vec_t base_offsets;
    out_ptr_vec_t out;

    #pragma unroll
    for (int i = 0; i < output_vec_size; i++) {
      base_offsets[i] = output_calc.get(output_idx + i)[0];
      out[i] = (out_scalar_t*)((char*)dst[0] + base_offsets[i]);
    }

    bool should_store = config.should_store(output_idx);
    if (should_store) {
      uint32_t offset = config.staging_memory_offset(blockIdx.y);
      reduce_buffer[offset] = value;
    }

    __threadfence(); // make sure writes are globally visible
    __syncthreads(); // if multiple warps in this block wrote to staging, make sure they're all done
    bool is_last_block_done = mark_block_finished();

    if (is_last_block_done) {
      value = ident;
      if (config.should_block_x_reduce()) {
        uint32_t input_offset = threadIdx.x + threadIdx.y * blockDim.x;
        uint32_t step = blockDim.x * blockDim.y;
        for (; input_offset < config.ctas_per_output; input_offset += step) {
          uint32_t idx = config.staging_memory_offset(input_offset);
          arg_vec_t next = reduce_buffer[idx];
          #pragma unroll
          for (int i = 0; i < output_vec_size; i++) {
            value[i] = reducer::combine(value[i], next[i]);
          }
        }
      } else {
        uint32_t input_offset = threadIdx.y;
        uint32_t step = blockDim.y;
        for (; input_offset < config.ctas_per_output; input_offset += step) {
          uint32_t idx = config.staging_memory_offset(input_offset);
          arg_vec_t next = reduce_buffer[idx];
          #pragma unroll
          for (int i = 0; i < output_vec_size; i++) {
            value[i] = reducer::combine(value[i], next[i]);
          }
        }
      }
      value = block_y_reduce(value, shared_memory);
      if (config.should_block_x_reduce()) {
        value = block_x_reduce<output_vec_size>(value, shared_memory);
      }
      if (should_store) {
        if (accumulate) {
          #pragma unroll
          for (int i = 0; i < output_vec_size; i++) {
            value[i] = reducer::translate_idx(value[i], base_idx);
          }
        }

        if (acc == nullptr) {
          if (accumulate) {
            value = accumulate_in_output<output_vec_size>(out, value);
          }
          if (final_output) {
            set_results_to_output<output_vec_size>(value, base_offsets);
          } else {
            #pragma unroll
            for (int i = 0; i < output_vec_size; i++) {
              *(out[i]) = get_accumulated_output(out[i], value[i]);
            }
          }
        } else {
          if (accumulate) {
            #pragma unroll
            for (int i = 0; i < output_vec_size; i++) {
              value[i] = reducer::combine((*acc)[i], value[i]);
            }
          }
          if (final_output) {
            set_results_to_output<output_vec_size>(value, base_offsets);
          } else {
            *acc = value;
          }
        }
      }
    }

    return value;
  }
}

extern "C"
__launch_bounds__(512, 4)
__global__ void reduction_prod_kernel(ReduceJitOp r){
  r.run();
}
nvrtc: error: invalid value for --gpu-architecture (-arch)

```

jdibenes, 2025-02-25:
I don't know how to fix that error, sorry.
But searching for `nvrtc: error: invalid value for --gpu-architecture (-arch)` suggests it may be CUDA or GPU driver incompatibility.
I'm using CUDA 12.3 on Windows 10.

sreeharshaparuchur1, 2025-02-25:
That's okay, thank you for your help thus far. I'm trying to run this on a Linux machine (ubuntu 22.04) with CUDA 12.4 so this error is puzzling to me too.

closed 2025-02-25 by sreeharshaparuchur1