jdibenes/hl2ss #161: VLC Camera's Extrinsicss Coordinate Frame Convention

frank-Regal, 2025-02-27:
Hello,
I want to use the camera extrinsics and the pose (given in the world frame) for each of the VLC cameras by this package and solve for the transform between the rignode of the Hololens and the HoloLens headset position

jdibenes, 2025-02-28:
Hi,

For the Research Mode sensors (including VLC), `data.pose` gives the pose of the rignode w.r.t. to the world.
For a given 3D point `p = [x; y; z; 1]` in rignode coordinates, `p.T @ data.pose` converts it to world coordinates.
For a 3D point in VLC coordinates, `p.T @ inv(calibration_vlc.extrinsics)` converts to rignode coordinates and the VLC to world conversion is `p.T @ inv(calibration_vlc.extrinsics) @ data.pose`.

The `HoloLens2SensorStreaming.cs` script links the coordinate system of the Unity scene to the hl2ss plugin, so the origin of the Unity scene (world) and hl2ss is the same. However, a right-to-left handed coordinate transform must be performed as described in https://learn.microsoft.com/en-us/windows/mixed-reality/develop/unity/unity-xrdevice-advanced?tabs=mrtk.
Then, if your $T_{HW}$ is left-handed and in conventional form, I think the transform should be `p.T @ data.pose @ diag(1,1,-1,1) @ T_{HW}.T` given `p` in rignode coordinates, but I'm not sure. 

We used to have a demo to test these things in https://github.com/jdibenes/hl2ss/blob/objload/viewer/unity_sample_twin_mesh.py to scan a room, upload the mesh to Unity on the HoloLens, and evaluate how the scan aligns with the real room. It might be useful.

frank-Regal, 2025-03-24:
@jdibenes 

Thank you for the details. This has helped me solve my problem. Out of curiosity, where are the extrinsics and intrinsics parameters obtained from? Is the hl2ss package accessing these from some sort of calibration file that is stored on the HoloLens from the factory?

jdibenes, 2025-03-24:
For the Research Mode sensors:

The Extrinsics are obtained directly using the Research Mode API
https://github.com/jdibenes/hl2ss/blob/8595431c24960ab5ff54e810c9997d4befc7f9b4/hl2ss/hl2ss/research_mode.cpp#L299-L319

For the cameras (VLC, Long Throw, AHAT), the images have some lens distortion so the Research Mode API does not give the intrinsic parameters ($K$) directly. Instead, it provides a method to convert image points to normalized coordinates (inverse $K$ if the images had no distortion) and another method to convert camera points to image points. Then, hl2ss uses these two methods to generate an image undistort map (preserving resolution and all valid pixels) and corresponding intrinsic parameters.
https://github.com/jdibenes/hl2ss/blob/8595431c24960ab5ff54e810c9997d4befc7f9b4/hl2ss/hl2ss/research_mode.cpp#L189-L296