jdibenes/hl2ss #163: Audio not streamed with sample_video.py

vli-syngular, 2025-03-17:
Hello, I'm testing with sample_video.py. All sensors data can be streamed except the audio. Am I missing something?

jdibenes, 2025-03-18:
Hi,
Audio playback is not implemented in sample_video.py.

vli-syngular, 2025-03-18:
Then is there any example demonstrating how to stream pv and extended audio at the same time?

jdibenes, 2025-03-18:
There wasn't but I just added an example in [sample_pv_audio.py](https://github.com/jdibenes/hl2ss/blob/9829aebd19c8d820d243424f4837fe74e6c4b5ef/viewer/sample_pv_audio.py#L1)

vli-syngular, 2025-03-18:
It works well! Thank you so much for adding this example.

vli-syngular, 2025-03-18:
Did you ever try using StreamingCaptureMode.AudioAndVideo to achieve that? Though I suppose your current solution is better in terms of performance.

jdibenes, 2025-03-18:
I tried it a long time ago. When using Exclusive capture, the audio controller would report 11 channels but the callback would not receive any audio frames so I didn't continue down that path.

vli-syngular, 2025-03-19:
Thanks for sharing those findings

vli-syngular, 2025-03-28:
Any idea how to eliminate the echo? I've tried changing MediaCategory::Meida to MediaCategory::Communications in c++ project, but then the microphone capture cannot be heard.