jdibenes/hl2ss #165: Eye Tracking accumulating lag

lucasbardis, 2025-03-20:
Hello, while displaying my gaze as a dot on the pv image, I noticed some latency. I used a reduced and modified version of "sample_eet_pv.py" to reproduce the issue. By comparing the timestamps, I found that the eye tracking acquisition is lagging behind the RGB acquisition.

I conducted several tests with different parameters (eye tracking FPS and acquisition methods such as 'nearest' or 'most_recent_frame') on multiple computers running both Ubuntu and Windows. While the intensity of the lag varies, it consistently occurs.

I'm running Python 3.11 and using the latest version of the library released three weeks ago. I'm utilizing the APPX created using the README because I use the RemoteUnityScene to display holograms (though not in this example).

I've included a sample code that reproduces the problem. Please let me know if you encounter the same issue. I typically let it run for 3 minutes at 30 FPS to observe the effect.

Thank you very much for your hardwork on this library !

[image]
[image]
[image]
[image]

```
from pynput import keyboard
import time
import multiprocessing as mp
import cv2
import hl2ss
import hl2ss_lnm
import hl2ss_mp
import matplotlib.pyplot as plt
import numpy as np

# HoloLens 2 address
host = "IP"

# Camera parameters
# See etc/hl2_capture_formats.txt for a list of supported formats
pv_width     = 960
pv_height    = 540
pv_framerate = 30

# EET parameters
eet_fps = 30 # 30, 60, 90

# Buffer length in seconds
buffer_length = 5

#------------------------------------------------------------------------------

if __name__ == '__main__':
    # Keyboard events ---------------------------------------------------------
    enable = True

    def on_press(key):
        global enable
        enable = key != keyboard.Key.esc
        return enable

    listener = keyboard.Listener(on_press=on_press)
    listener.start()

    # Start PV Subsystem ------------------------------------------------------
    hl2ss_lnm.start_subsystem_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO)

    # Start PV and EET streams ------------------------------------------------
    producer = hl2ss_mp.producer()
    producer.configure(hl2ss.StreamPort.PERSONAL_VIDEO, hl2ss_lnm.rx_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO, width=pv_width, height=pv_height, framerate=pv_framerate))
    producer.configure(hl2ss.StreamPort.EXTENDED_EYE_TRACKER, hl2ss_lnm.rx_eet(host, hl2ss.StreamPort.EXTENDED_EYE_TRACKER, fps=eet_fps))
    producer.initialize(hl2ss.StreamPort.PERSONAL_VIDEO, pv_framerate * buffer_length)
    producer.initialize(hl2ss.StreamPort.EXTENDED_EYE_TRACKER, hl2ss.Parameters_SI.SAMPLE_RATE * buffer_length)
    producer.start(hl2ss.StreamPort.PERSONAL_VIDEO)
    producer.start(hl2ss.StreamPort.EXTENDED_EYE_TRACKER)

    consumer = hl2ss_mp.consumer()
    manager = mp.Manager()
    sink_pv = consumer.create_sink(producer, hl2ss.StreamPort.PERSONAL_VIDEO, manager, ...)
    sink_eet = consumer.create_sink(producer, hl2ss.StreamPort.EXTENDED_EYE_TRACKER, manager, None)
    sink_pv.get_attach_response()
    sink_eet.get_attach_response()

    t_elapsed_array = []
    lag = []
    first_timestamp = -1
    acquire_from = "most_recent_frame"
    # acquire_from = "nearest"
    frame = 0

    # Main Loop ---------------------------------------------------------------
    while (enable):
        t_start_frame = time.time()
        # Wait for PV frame ---------------------------------------------------
        sink_pv.acquire()

        # Get PV frame and nearest (in time) EET frame ------------------------
        _, data_pv = sink_pv.get_most_recent_frame()
        if ((data_pv is None) or (not hl2ss.is_valid_pose(data_pv.pose))):
            continue

        if acquire_from == "nearest":
            _, data_eet = sink_eet.get_nearest(data_pv.timestamp)
        elif acquire_from == "most_recent_frame":
            _, data_eet = sink_eet.get_most_recent_frame()
        if ((data_eet is None) or (not hl2ss.is_valid_pose(data_eet.pose))):
            continue

        image = data_pv.payload.image
        eet = hl2ss.unpack_eet(data_eet.payload)

        if first_timestamp == -1:
            first_timestamp = data_pv.timestamp / 10000000
        rgb_timestamp = (
            data_pv.timestamp / 10000000
        ) - first_timestamp
        et_timestamp = (
            data_eet.timestamp / 10000000
        ) - first_timestamp
        lag.append(rgb_timestamp - et_timestamp)

        # Display frame -------------------------------------------------------            
        # cv2.imshow('Video', image)
        # cv2.waitKey(1)
        if frame == 5400:
            enable = False
        t_elapsed_frame = time.time() - t_start_frame
        if len(t_elapsed_array) < 100:
            t_elapsed_array.append(t_elapsed_frame)
        else:
            t_elapsed_array.pop(0)
            t_elapsed_array.append(t_elapsed_frame)
            if frame % 60 == 0:
                print(
                    f"Frame {frame} : {np.round(1 / np.mean(t_elapsed_array), 1)} fps"
                )
        frame += 1

    # Stop PV and EET streams -------------------------------------------------
    sink_pv.detach()
    sink_eet.detach()
    producer.stop(hl2ss.StreamPort.PERSONAL_VIDEO)
    producer.stop(hl2ss.StreamPort.EXTENDED_EYE_TRACKER)

    # Stop PV subsystem -------------------------------------------------------
    hl2ss_lnm.stop_subsystem_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO)

    # Stop keyboard events ----------------------------------------------------
    listener.join()

    plt.plot(lag)
    plt.title(f"Lag {acquire_from} with {eet_fps} FPS")
    plt.xlabel("Frames")
    plt.ylabel("Time (s)")
    plt.show()
```

jdibenes, 2025-03-21:
Hello,

Thank you for your report. The graphs and script are very helpful.
I can confirm the issue happens on my end too.
It seems the hl2ss_mp extension is just too slow for the extended eye tracker receiver.

As a workaround, you could try using the hl2ss_mt extension ([client_python_mt](https://github.com/jdibenes/hl2ss/tree/main/extensions)).
Here's a graph using the hl2ss_mt extension:
[image]

This is the script adjusted to use hl2ss_mt:
```python
from pynput import keyboard
import time
import cv2
import hl2ss
import hl2ss_lnm
import hl2ss_mt
import matplotlib.pyplot as plt
import numpy as np

# HoloLens 2 address
host = "192.168.1.7"

# Camera parameters
# See etc/hl2_capture_formats.txt for a list of supported formats
pv_width     = 960
pv_height    = 540
pv_framerate = 30

# EET parameters
eet_fps = 90 # 30, 60, 90

# Buffer length in seconds
buffer_length = 5

#------------------------------------------------------------------------------

if __name__ == '__main__':
    # Keyboard events ---------------------------------------------------------
    enable = True

    def on_press(key):
        global enable
        enable = key != keyboard.Key.esc
        return enable

    listener = keyboard.Listener(on_press=on_press)
    listener.start()

    # Start PV Subsystem ------------------------------------------------------
    hl2ss_lnm.start_subsystem_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO)

    # Start PV and EET streams ------------------------------------------------
    cfg_pv = hl2ss_mt.create_configuration(hl2ss.StreamPort.PERSONAL_VIDEO)
    cfg_eet = hl2ss_mt.create_configuration(hl2ss.StreamPort.EXTENDED_EYE_TRACKER)

    cfg_pv['width'] = pv_width
    cfg_pv['height'] = pv_height
    cfg_pv['framerate'] = pv_framerate
    cfg_eet['framerate'] = eet_fps

    sink_eet = hl2ss_mt.rx_eet(host, hl2ss.StreamPort.EXTENDED_EYE_TRACKER, eet_fps * buffer_length, cfg_eet)
    sink_pv = hl2ss_mt.rx_decoded_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO, pv_framerate * buffer_length, cfg_pv)

    sink_eet.open()
    sink_pv.open()

    t_elapsed_array = []
    lag = []
    first_timestamp = -1
    #acquire_from = "most_recent_frame"
    acquire_from = "nearest"
    frame = 0

    pv_index = -1

    # Main Loop ---------------------------------------------------------------
    while (enable):
        t_start_frame = time.time()
        # Wait for PV frame ---------------------------------------------------

        # Get PV frame and nearest (in time) EET frame ------------------------
        while (True):
            data_pv = sink_pv.get_by_index(pv_index)
            if (data_pv.status == hl2ss_mt.Status.WAIT):
                time.sleep(1/1000)
                continue
            elif (data_pv.status == hl2ss_mt.Status.DISCARDED):
                pv_index = -1
                continue
            break

        pv_index = data_pv.frame_stamp + 1
            
        if acquire_from == "nearest":
            data_eet = sink_eet.get_by_timestamp(data_pv.timestamp)
        elif acquire_from == "most_recent_frame":
            data_eet = sink_eet.get_by_index(-1)
        if ((not hl2ss.is_valid_pose(data_pv.pose)) or (data_eet is None) or (not hl2ss.is_valid_pose(data_eet.pose))):
            time.sleep(1/1000)
            continue

        image = data_pv.payload.image
        eet = hl2ss.unpack_eet(data_eet.payload)

        if first_timestamp == -1:
            first_timestamp = data_pv.timestamp / 10000000
        rgb_timestamp = (
            data_pv.timestamp / 10000000
        ) - first_timestamp
        et_timestamp = (
            data_eet.timestamp / 10000000
        ) - first_timestamp
        lag.append(rgb_timestamp - et_timestamp)

        # Display frame -------------------------------------------------------            
        # cv2.imshow('Video', image)
        # cv2.waitKey(1)
        if frame == 5400:
            enable = False
        t_elapsed_frame = time.time() - t_start_frame
        if len(t_elapsed_array) < 100:
            t_elapsed_array.append(t_elapsed_frame)
        else:
            t_elapsed_array.pop(0)
            t_elapsed_array.append(t_elapsed_frame)
            if frame % 60 == 0:
                print(
                    f"Frame {frame} : {np.round(1 / np.mean(t_elapsed_array), 1)} fps"
                )
        frame += 1

    # Stop PV and EET streams -------------------------------------------------
    sink_pv.close()
    sink_eet.close()

    # Stop PV subsystem -------------------------------------------------------
    hl2ss_lnm.stop_subsystem_pv(host, hl2ss.StreamPort.PERSONAL_VIDEO)

    # Stop keyboard events ----------------------------------------------------
    listener.join()

    plt.plot(lag)
    plt.title(f"Lag {acquire_from} with {eet_fps} FPS")
    plt.xlabel("Frames")
    plt.ylabel("Time (s)")
    plt.show()
```

jdibenes, 2025-03-24:
Extended Video uses the same objects as PV, see [client_xmt_extended_video.py](https://github.com/jdibenes/hl2ss/blob/main/viewer/client_xmt_extended_video.py).

closed 2025-03-25 by lucasbardis